openhwfoundation/cva6 #1222: execute work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf

Shuangxie0708, 2023-05-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When I running User-Space Applications..I execute work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf There is below error that failed.
But I run work-ver/Variane_testharness rv64um-v-divuw, it is successful.
Could you help me solve this problem that why I failed when running user-space applications. Thanks!

**root@samsung-Precision-5820-Tower-X-Series:/home/samsung/riscv/cva6# work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf**
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 38919
*** [rvf_tracer] WARNING: No valid address of 'tohost' (tohost == 0x00000000000000), termination possible only by timeout or Ctrl-C!

bbl loader

**/home/samsung/riscv/install_riscv/riscv64-unknown-elf/bin/pk *** FAILED *** (tohost = 2147483647) after 2000012 cycles**

CPU time used: 256327.08 ms
Wall clock time passed: 256427.19 ms
root@samsung-Precision-5820-Tower-X-Series:/home/samsung/riscv/cva6#

JeanRochCoulon, 2023-05-12:
@Shuangxie0708 uses pk, @eyssartk as you are the closest person I know to the issue, I assigned it to you. Do not hesitate to forward the "hot potato" if you think somebody else could help. The objective is to find the right person to give feedback !

retina007, 2023-05-23:
I try to booting Linux-64 CV64A6 using Verilator or Questa simulator. is this the command to use?（work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf）

Shuangxie0708, 2023-05-23:
yes, you need use the command (work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf)

rishikakushwah16, 2023-06-12:
Hey @Shuangxie0708 , I am having the same issue, is there any solution for the problem?  work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf 
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 40083
*** [rvf_tracer] WARNING: No valid address of 'tohost' (tohost == 0x00000000000000), termination possible only by timeout or Ctrl-C!

bbl loader
/home/rkushwah/riscv-tools/riscv64-unknown-elf/bin/pk *** FAILED *** (tohost = 2147483647) after 2000012 cycles
CPU time used: 471939.98 ms
Wall clock time passed: 471956.68 ms

yulong-lan, 2023-07-08:
I am wondering where this "FAILED" message comes from, which I have also got. 

Because when you pass another command -v hello.vcd in between to dump a vcd file, as below: 
`work-ver/Variane_testharness -v hello.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf`

It does generate a non-empty hello.vcd file, and open it you will see all the signals, which indicates that the programme has been actually running. 

Beware though! Notwithstanding a very tiny "hello" programme, the size of vcd file ends up more than 15 GB, I had to terminate the simulation halfway since it was completely running up my disk space.

allensted, 2023-10-14:
try to use puts() instread of printf()， that works for me.

i guess the env or the core or maybe the pk can not support printf() well.

int main() {
    //printf("Hello CVA6!\\n");
    puts("Hello CVA6!\\n");
    return 0;
}

JeanRochCoulon, 2023-10-24:
Related to #1539

tjt7a, 2023-10-30:
I run into the same problem; I think the issue is that the simulator won't simulate beyond 2000012 cycles, even after I set the max_cycles parameter in the makefile:
# Maximum amount of cycles for a successful simulation run
max_cycles     ?= 100000000

JeanRochCoulon, 2023-10-31:
Indeed, there is a RTL timeout implemented in rvfi_tracer.sv in SIM_FINISH variable.

JeanRochCoulon, 2024-02-14:
As the discussion seems to be over, I will close the issue.

closed 2024-02-14 by JeanRochCoulon