openhwfoundation/cva6 #1237: [BUG] <Unknown Values on performance counters>

AyoubJalali, 2023-05-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
The problem here is the a got unknown values on the performance counters (**mhpmcounter-h & mhpmevent**). 

_core   0: 0x0000000080000b0c (0xb1e792f3) csrrw   t0, mhpmcounter30, a5
3 0x0000000080000b0c (0xb1e792f3) x 5 0xxxxxxxxx_ (example)

so to reproduce the problem you have to : 

- Add the yaml description of the test in [cva6_base_testlist.yaml](https://github.com/openhwgroup/core-v-verif/blob/master/cva6/sim/cva6_base_testlist.yaml)

- test: riscv_arithmetic_basic_hint_illegal_csr_test
  description: >
    Arithmetic instruction test, no load/store/branch instructions
  gen_opts: >
    +instr_cnt=500
    +num_of_sub_program=5
    +directed_instr_0=riscv_int_numeric_corner_stream,4
    +no_data_page=1
    +no_branch_jump=1
    +boot_mode=m
    +no_csr_instr=0
    +enable_illegal_csr_instruction=1
    +hint_instr_ratio=100
    +tvec_alignment=8
  iterations: 2
  gen_test: riscv_instr_base_test
  rtl_test: core_base_test

and run the commend : 

1. python3 cva6.py --testlist=cva6_base_testlist.yaml --test riscv_arithmetic_basic_hint_illegal_csr_test --iss_yaml cva6.yaml --target cv32a60x --iss=vcs-uvm,spike -i 1 --iss_timeout 300 --seed 3018203365 (bug on mhpmcounter12h)

2. python3 cva6.py --testlist=cva6_base_testlist.yaml --test riscv_arithmetic_basic_hint_illegal_csr_test --iss_yaml cva6.yaml --target cv32a60x --iss=vcs-uvm,spike -i 1 --iss_timeout 300 --seed 2951308182 (bug on mhpmcounter15)

3. python3 cva6.py --testlist=cva6_base_testlist.yaml --test riscv_arithmetic_basic_hint_illegal_csr_test --iss_yaml cva6.yaml --target cv32a60x --iss=vcs-uvm,spike -i 1 --iss_timeout 300 --seed 1147271990 (bug on mhpmevent8)

in core-v-verif/cva6/sim (branch cva6/dev & RTL matser), definitely you gonna find one of test failed because of an unknown values on these perform counter without seeds if you want to.

TulikaSi, 2023-06-01:
Hi @AyoubJalali 
Just a quick question
Did you do any other change apart from the one in the yaml file?

AyoubJalali, 2023-06-01:
> Did you do any other change apart from the one in the yaml file?

Yes As i remember

AyoubJalali, 2023-06-01:
> It will be good if you could let me know about that as well

you having some problems to regenerate the bug ?

TulikaSi, 2023-06-01:
This is what we are getting 

[image]

Kindly let us know if we are going wrong somewhere

AyoubJalali, 2023-06-01:
Ahh Yes, you're in the wrong version of RISCV-DV : 
go to this version of RISCV-DV (cva6/sim/dv/ 0ce85d3)

JeanRochCoulon, 2023-07-10:
@TulikaSi   Did you succeed in regenerating the issue ?

TulikaSi, 2023-08-08:
@AyoubJalali @JeanRochCoulon @ASintzoff 
Can we close this issue now??

AyoubJalali, 2023-08-08:
We can, let's wait until the PR merged

closed 2023-08-31 by AyoubJalali