openhwfoundation/cva6 #1243: Choose the number of PMP entries in CVA6 processor

KevinQhv, 2023-05-23:
Hello ! 

I try to choose the number of PMP entries as indicated in the  documentation [https://cva6.readthedocs.io/en/latest/01_cva6_user/PMP.html](url) by manipulating the variable `ArianeCfg.NrPMPEntries`. This variable is in several files and I change the number of PMP entries and compare between 0 PMP entries and 16 PMP entries. But I get the same number of LUT, Flip Flop, ...

Do I change the number of PMP entries correctly ?

Thank you in advance

KevinQhv, 2023-05-30:
I found the answer to my problem. It was in the documentation, in another section _[Execute Stage - PMA/PMP Checks
(https://cva6.readthedocs.io/en/latest/03_cva6_design/ex_stage.html#pma-pmp-checks)_.
it says : _"Only 8 PMPs are implemented. This simplifies implementation as we don't have to worry about unaligned accesses."_

Moschn, 2023-05-30:
I think this is incorrect. We only support granularity 8, but the number of PMP entries should be configurable. This is a valid bug. I would propose to re-open and assign to someone that can fix this.

KevinQhv, 2023-06-29:
The number of PMP entries can be configured with the variable `CVA6ConfigNrPMPEntries`. 
In the `cva6/core/include/ariane_pkg.sv` file, PMP entries are configured with the data `NrPMPEntries`: `unsigned'(cva6_config_pkg::CVA6ConfigNrPMPEntries) ` coming from a CVA6 processor configuration file such as : 

`cv..a6_...._config_pkg.sv ` where you can find the variable to be modified :
`localparam CVA6ConfigNrPMPEntries = 8;` default is 8 entries.

closed 2023-06-29 by KevinQhv