openhwfoundation/cva6 #1250: [BUG] <title>Using a third-party debugger, how do I modify /fpga/ariane.cfg to debug normally?

StilluSmile, 2023-05-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I have transplanted CVA6 to a non-genesys2 FPGA development board, and I can also use the sd card to start the linux system. Now I want to use openocd to debug on FPGA, and the pin of jtag has been mapped to the GPIO of my FPGA development board. The current problem is that I will use a third-party debugger for debugging. After the debugger is connected, enter lsusb in the terminal and print as follows:

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 013: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC
Bus 003 Device 004: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 012: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 003 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 009: ID 32bf:b210 C-Sky MicroSystem Co., Ltd. C-Sky CKLink-Lite
Bus 003 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Bus 003 Device 009: ID 32bf:b210 C-Sky MicroSystem Co., Ltd. C-Sky CKLink-Lite     This one is my debugger id

How do I modify /fpga/ariane.cfg to debug normally?

Thank you very much for any replies and suggestions~

StilluSmile, 2023-05-31:
If I type openocd -f ./ariane.cfg directly on the terminal, the output is as follows:

Open On-Chip Debugger 0.11.0+dev-02383-g8bb25e007 (2022-09-27-14:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'ftdi layout_signal' not 'ftdi_layout_signal'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'

eyssartk, 2023-05-31:
Have you try changing 

ftdi_vid_pid 0x0403 0x6010
at https://github.com/openhwgroup/cva6/blob/master/corev_apu/fpga/ariane.cfg#L4 ?

StilluSmile, 2023-06-01:
> Have you try changing
> 
> ftdi_vid_pid 0x0403 0x6010 at https://github.com/openhwgroup/cva6/blob/master/corev_apu/fpga/ariane.cfg#L4 ?

Thank you for your reply @eyssartk 

I've tried replacing ftdi_vid_pid in ariane.cfg with the id 32bf:b210  of the third party debugger.

 #ftdi_vid_pid 0x0403 0x6010
 ftdi_vid_pid 0x32bf 0xb210

 Also, add the following udev rule to /etc/udev/rules.d/99-ftdi.rules:
 SUBSYSTEM=="usb", ACTION=="add", ATTRS{idProduct}=="b210", ATTRS{idVendor}=="32bf", MODE="666", GROUP="plugdev"

When I try openocd -f fpga/ariane.cfg again, it still doesn't work, the output is as follows:

Open On-Chip Debugger 0.11.0+dev-02383-g8bb25e007 (2022-09-27-14:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'ftdi layout_signal' not 'ftdi_layout_signal'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
Error: unsupported FTDI chip type: 0x0212
Error: unable to open ftdi device with vid 32bf, pid b210, description '*', serial '*' at bus location '*'

It seems that the FTDI chip is not supported, maybe other configurations in ariane.cfg need to be modified?

zarubaf, 2023-10-23:
Unfortunately, we can't support all available debuggers. This question might be better asked in the OpenOCD issue tracker?