openhwfoundation/cva6 #1253: [BUG] openocd still doesn't work

StilluSmile, 2023-06-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I have transplanted CVA6 to a non-genesys2 FPGA development board, and I can also use the sd card to start the linux system. Now I want to use openocd to debug on FPGA, and the pin of jtag has been mapped to the GPIO of my FPGA development board. Since my board does not have a debug connection environment like the genesys2 board, I repurchased a Nuclei debugger, which uses the FTDI 2232 chip inside it. After the debugger is connected, enter lsusb in the terminal and print as follows:

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 007: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Bus 003 Device 004: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

It seems to be consistent with the ID described in the debug section.After entering lsusb, the FTDI chip information is also printed.

Bus 003 Device 012: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC

When I try openocd -f fpga/ariane.cfg again, it still doesn't work, the output is as follows:

Open On-Chip Debugger 0.11.0+dev-02383-g8bb25e007 (2022-09-27-14:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'ftdi layout_signal' not 'ftdi_layout_signal'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : [riscv.cpu] datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112d
[riscv.cpu] Target successfully examined.
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

It seems that it stopped at the line "Info : Listening on port 4444 for telnet connections", and did not print out "Info : accepting 'gdb' connection on tcp/3333" as in the routine.

It may need to be modified in ariane.cfg or other configuration files. How to modify the configuration to debug normally?

Thank you very much for any replies and suggestions~

zarubaf, 2023-06-05:
It literally reads:
```
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
```

have you tried connecting with GDB on that port?

StilluSmile, 2023-06-05:
> 
Thank you for your reply @zarubaf 

I use openocd for the first time, which port do you mean to connect? It means entering riscv64-unknown-elf-gdb ./hello.elf in a new terminal window
After entering, the prompt is as follows:

GNU gdb (SiFive GDB 8.3.0-2020.04.0) 8.3
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/sifive/freedom-tools/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) load hello.elf 
You can't do that when your target is `None'
(gdb)

zarubaf, 2023-06-05:
After starting GDB you can do `target remote localist:3333` to connect to the debug bridge. Then you’ll be able to load the program, and debug it.

StilluSmile, 2023-06-05:
> target remote localhost:3333

ok，i will try it. tell u result later~

StilluSmile, 2023-06-05:
after entering 

> > target remote localhost:3333
> 
> ok，i will try it. tell u result later~
thank u very much，@zarubaf Now I can connect to gdb and load my program, as shown below. 

(gdb) load hello.elf 
Loading section .text, size 0x2532 lma 0x100b0
Loading section .rodata, size 0x18 lma 0x125e8
Loading section .eh_frame, size 0x4 lma 0x13600
Loading section .init_array, size 0x10 lma 0x13608
Loading section .fini_array, size 0x8 lma 0x13618
Loading section .data, size 0xf58 lma 0x13620
Loading section .sdata, size 0x40 lma 0x14578
Start address 0x100c8, load size 13566
Transfer rate: 47 KB/sec, 1938 bytes/write.
(gdb) pwd 
Working directory /home/cva6/tmp/riscv-tests/build.
(gdb) load ./benchmarks/dhrystone.riscv
Loading section .text.init, size 0x1c6 lma 0x80000000
Loading section .tohost, size 0x48 lma 0x80001000
Loading section .text, size 0x96c lma 0x80002000
Loading section .text.startup, size 0x6fe lma 0x8000296c
Loading section .rodata.str1.8, size 0x668 lma 0x80003070
Loading section .rodata, size 0x158 lma 0x800036d8
Start address 0x80000000, load size 6712
Transfer rate: 31 KB/sec, 1118 bytes/write.
(gdb) c
Continuing.

However, I know that the previous hello.elf cannot be run, because the link address is wrong, it should be mapped from 0x80000000. I tried to load the compiled program in the benchmark folder and found that it still couldn't run. So the question now is how to get a linker script so that the compiled program can be loaded and run.

retina007, 2023-06-07:
@StilluSmile I will use  Nuclei debugger and other fpga board.   can contact me with wdlong007@gmail.com

StilluSmile, 2023-06-07:
> 

thank u for your reply~ @retina007 
do u encounter above problem when u port cva6 to other fpga boards and debug with openocd?
Emm... maybe  it would be better that we discuss our questions here ~

zarubaf, 2023-06-07:
The execution looks good, but probably the `printf` is based on `fesvr`, and you don’t have the frontend server available on the FPGA. Since you didn’t boot an OS, you need to proved the UART driver yourself. You can take inspiration / copy from the bootrom folder.  I know that this isn’t super helpful, and it would be better if would provide a working example.

StilluSmile, 2023-06-08:
> The execution looks good, but probably the `printf` is based on `fesvr`, and you don’t have the frontend server available on the FPGA. Since you didn’t boot an OS, you need to provide the UART driver yourself. You can take inspiration / copy from the bootrom folder. I know that this isn’t super helpful, and it would be better if would provide a working example.

thank u very much @zarubaf

Actually, when I was debugging with gdb, I started the linux system. Do you mean the driver that needs to provide serial port printing? If so, where is this driver placed? It would be very convenient to provide such an example for those interested in cva6. hhh

zarubaf, 2023-10-23:
There is a bare-metal library available here: https://github.com/ThalesGroup/cva6-baremetal-bsp
With Linux you should get the necessary console drivers. Closing this issue now, feel free to re-open in case the issue wasn't addressed properly.

closed 2023-10-23 by zarubaf

chenqian13, 2023-12-20:
Hi! I want to know the content of the file fpga/ariane.cfg that you used.  Also, I would like to know what .rules files are in your/etc/udev/rules.d directory and what their contents are.
Could you please tell me the information above? Thanks a lot!!!

chenqian13, 2023-12-20:
@StilluSmile Hi! I want to know the content of the file fpga/ariane.cfg that you used. Also, I would like to know what .rules files are in your/etc/udev/rules.d directory and what their contents are.
Could you please tell me the information above? Thanks a lot!!!

StilluSmile, 2023-12-21:
> @StilluSmile Hi! I want to know the content of the file fpga/ariane.cfg that you used. Also, I would like to know what .rules files are in your/etc/udev/rules.d directory and what their contents are. Could you please tell me the information above? Thanks a lot!!!

Hi~, @chenqian13. I use the original cfg configuration under the fpga directory.  As for the .rules file, we still use the content described in the Debugging section of the README.