openhwfoundation/cva6 #1260: Issue with SIE CSR Access modes

spidugu444, 2023-06-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

HI,

While verifying Access mode for SIE(Supervisor Interrupt enable) CSR on CVA6 core, test got failed due to CSR is always reading Zero on both spike  and verilator.
As per CVA6 user manual please find the access modes for SIE CSR is as mentioned below, 

[image]

MikeOpenHWGroup, 2023-08-23:
Hi @spidugu444, in the test-program that tests the access mode behavior of the SIE CSR, do you set values of the `MIDELEG` CSR?  I believe if you want to write non-zero to MIE or SIE, you need to set the appropriate bits in MIDELEG.

spidugu444, 2023-08-30:
@MikeOpenHWGroup, by configuring MIDELEG CSRs supervisor Interrupt bits, now I am able to verify access modes for SIE CSR.

spidugu444, 2023-08-30:
As I am able to access SIE CSR, I am closing this issue.

closed 2023-08-30 by spidugu444