openhwfoundation/cva6 #1261: Issue with the privilege mode for CYCLE and INSTRET CSRs

spidugu444, 2023-06-11:
HI,

In RISC-V privileged specification the `cycle` and `instret` CSRs are marked as User Read-only (URO). This means they can be accessed in User mode as read-only registers. They can also be accessed in higher privilege modes (Supervisor and Machine mode) as read-only.

While verifying access modes for user mode CSRs **CYCLE and INSTRET** on CVA6 core, these are raising illegal instruction exception on Supervisor mode and User mode.

could anyone please let me know that whether we can access CYCLE and INSTRET CSRs on Supervisor and User mode.

spidugu444, 2023-06-11:
@RanjanThales

TulikaSi, 2023-06-19:
Hi @spidugu444 
In order to read the **cycle** and **instret** CSRs in S-mode and U-mode you will need to set the the corresponding bits in **mcounteren**  and **scounteren** register else the access will cause illegal instruction exception.

spidugu444, 2023-06-19:
@TulikaSi, Thank you for your response. I was able to enter the USER mode cycle and insert CSRs in M, S, and U mode after setting the SCOUNTEREN and MCOUNTEREN CSRs to 0x7.

closed 2023-06-19 by spidugu444