openhwfoundation/cva6 #1279: Setup.sh to configure verification environment

JeanRochCoulon, 2023-06-23:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Hello,

A script of installation is needed to well configure the verif env.

A first example could be the following one, but this need to be updated to support generic path  

# setup
export SPIKE_INSTALL_DIR="__local__"
export ROOT_PROJECT=<path_to>/core-v-verif
export  RISCV=<path_to>/gcc-10.2.0
export  VERILATOR_INSTALL_DIR=<path_to>/verilator-5.008
export  SPIKE_ROOT=<path_to>/spike
export  BBL_ROOT=<path_to>/Linux-ariane-sdk
source <path_to>/S-2021.09-SP1-1/setup/bashrc.example
rm -rf vendor/riscv/riscv-isa-sim/build
source cva6/regress/install-spike.sh |& tee spike-build.log
sh cva6/regress/smoke-tests.sh

Regards
Jean-Roch

### Description of Done

Starting point

zchamski, 2023-06-23:
The configuration script should take into account two "static" aspects of configuration:
* site-*dependent* configuration (location of compilers, digital design tools, ...)
* site-*independent* core-v-verif configuration (default location of vendorized Spike, locally built Verilator etc.).

The dynamic, per-run configuration such as the choice of HW/ISS target, compiler version and compile-time ISA extensions should be managed outside the setup script.

JeanRochCoulon, 2023-10-24:
Related to #1569

valentinThomazic, 2023-12-27:
Related to https://github.com/openhwgroup/cva6/issues/1727

valentinThomazic, 2024-01-05:
@JeanRochCoulon I think this issue can be closed since this has been implemented.

JeanRochCoulon, 2024-01-05:
Hello @valentinThomazic !!  Yes I close it

closed 2024-01-05 by JeanRochCoulon