openhwfoundation/cva6 #1300: [BUG] About four-byte alignment on target addresses of control transfer instructions (branch_unit.sv)

yaohsiaopid, 2023-07-10:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,
 
In `core/branch_unit.sv` the `branch_exception_o.valid` is raised high only if the branch/jump target address is not aligned **on a two-byte boundary** ([line 94](https://github.com/openhwgroup/cva6/blob/513bb91f823a18a6653a1685acd60408049c9c8f/core/branch_unit.sv#L94C8-L96C10) checks only the LSB). That is, it enforces only two-byte boundary check, and does not raise exception when target address of JAL/JALR is not four-byte aligned. 
However, in RISC-V 2.1 specification p.16 says "The JAL and JALR instructions can generate a misaligned instruction fetch exception if the target address is not aligned to a four-byte boundary."

If my understanding is correct, I think the fix may need to change the condition of [line 94](https://github.com/openhwgroup/cva6/blob/513bb91f823a18a6653a1685acd60408049c9c8f/core/branch_unit.sv#L94C8-L96C10) from. 
`if (branch_valid_i && target_address[0] != 1'b0) branch_exception_o.valid = 1'b1;` to   
`if (branch_valid_i && (target_address[1:0] != 2'b00)) branch_exception_o.valid = 1'b1;`

Thanks!

JeanRochCoulon, 2023-08-28:
@yaohsiaopid   Thank you for this bug finding. Can you submit a PR to fix it ?

JeanRochCoulon, 2023-12-10:
I submitted #1690

cathales, 2023-12-11:
I think the elaboration condition for C extension should be considered in the condition:

> Instruction-address-misaligned exceptions are not possible on machines that support extensions
with 16-bit aligned instructions, such as the compressed instruction-set extension, C.

Volume I: RISC-V Unprivileged ISA V20191213, page 21

PR #1719 (2023-12-19): fix: exception on misaligned branch if no RVC
The bit at index 1 must be 0 if the RVC extension is disabled.

Fix #1300

Replace PR #1690
closed 2023-12-19 by JeanRochCoulon