openhwfoundation/cva6 #1301: [BUG] Infinite loop in std_cache_pkg::one_hot_to_bin()

maxbjurling, 2023-07-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

If the `in` input to the [one_hot_to_bin()](https://github.com/openhwgroup/cva6/blob/513bb91f823a18a6653a1685acd60408049c9c8f/core/include/std_cache_pkg.sv#L78) function is 0 and  `ariane_pkg::DCACHE_SET_ASSOC` is a power of 2 then this function enters an infinite loop since the loop variable`i` wraps from DCACHE_SET_ASSOC-1 to 0 and the loop condition is always true.

Changing the type of `i` to `int unsigned` resolves this issue.

maxbjurling, 2023-07-11:
@fatimasaleem I think this was caused by your pull request https://github.com/openhwgroup/cva6/pull/1297

fatimasaleem, 2023-07-11:
> @fatimasaleem I think this was caused by your pull request #1297

Thank you @maxbjurling for pointing this out, I will try to resolve this lint warning with a different approach.

niwis, 2023-07-11:
There is a common cell for this:
https://github.com/pulp-platform/common_cells/blob/master/src/onehot_to_bin.sv
 Maybe replacing `std_cache_pkg::one_hot_to_bin()` with this might be a solution?

maxbjurling, 2023-07-11:
@niwis Reverting the single line that was changed in #1297 seems more straightforward than replacing a function call with a module instantiation. At least as an initial solution.

JeanRochCoulon, 2023-08-28:
@maxbjurling    Can you confirm the bug is fixed and close this issue ?   Thank you BTW for your finding

maxbjurling, 2023-08-29:
Bug is fixed. Thank you!

closed 2023-08-29 by maxbjurling