openhwfoundation/cva6 #1312: Issue with MTVEC CSR default value

spidugu444, 2023-07-20:
Hi,

I was working on CVA6 CSR access mode verification and I found that as per CVA6 user manual reset value for MTVEC CSR is zero, but in CVA6 RTL I am reading default value as 0x10040.

Following is a snap of [csr_regfile.sv](https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv) file where we can see the value of mtvec being assigned which comes out to be 0x10040

        // check whether we come out of reset
        // this is a workaround. some tools have issues
        // having boot_addr_i in the asynchronous
        // reset assignment to mtvec_d, even though
        // boot_addr_i will be assigned a constant
        // on the top-level.
        if (mtvec_rst_load_q) begin
            mtvec_d             = {{riscv::XLEN-riscv::VLEN{1'b0}}, boot_addr_i} + 'h40;
        end else begin
            mtvec_d             = mtvec_q;
        end

Can I know what should be the correct reset value for the mtvec CSR or the explanation for the above value of the mtvec CSR (0x10040)??

Thanks

spidugu444, 2023-07-20:
@RanjanThales

MikeOpenHWGroup, 2023-08-23:
Hi @frikhaAziz, can you investigate this issue?

MikeOpenHWGroup, 2023-08-23:
Hi @ASintzoff.  @JeanRochCoulon let me know that @frikhaAziz has completed his internship, so I am assigning this one to you.

ASintzoff, 2023-11-13:
The CSR reset values in https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CV32A6_Control_Status_Registers.rst were provided by Jade tool. This file needs to be updated with IPXACT generated file.

JeanRochCoulon, 2025-01-02:
@zchamski We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case ?

zchamski, 2025-01-09:
We need to align the spec and the RTL (both are wrong):

* old CSR spec from IP-XACT for CV32A60X states that `mtvec` reset value is `0x0`;
* the CV32A65X `riscv-config` spec states that `mtvec` reset value is `0x80001000`;
* the RTL sets `mtvec_d` to `boot_addr_i + 0x40`  upon coming out of  reset, but during reset `mtvec_q` is held at `0x0`.
* Spike initializes `mtvec` with `0x0`.

JeanRochCoulon, 2025-01-09:
As MTVEC value is setup by software during boot, never mind its hardware reset value. The best is to reset at 0. riscv-config need to be updated, and CSR specification regenerated.

zchamski, 2025-01-23:
I wiill update the riscv-config and the files derived from it.

JeanRochCoulon, 2025-01-24:
Superseeded by #2738

closed 2025-01-24 by JeanRochCoulon