openhwfoundation/cva6 #1332: unnecessary logic

omerguzelelectronicguy, 2023-08-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

This cannot be counted as bug but I couldn't find a relevant label. The tools already handle it. However, It is better to simple as much as possible. That's why I wanted to ask whether[ this line ](https://github.com/openhwgroup/cva6/blob/a052d4f2d54061e788924f7f3d1fa8e5933c0980/core/csr_regfile.sv#L873)includes unnecessary logic or not. 

Let say `!pmpcfg_q[index].locked` `a`, and `pmpcfg_q[index].locked && pmpcfg_q[index].addr_mode == riscv::TOR` `b`.
the if statement depend on `a.(a'.b)'`
`= a.(a+b')`
Which is `a` so the actual if statement must depend on `!pmpcfg_q[index].locked`

Moschn, 2023-08-14:
Thanks for the report! I think there might be a `+1` missing. According to the comment, it should be: 

```systemverilog
// check if the entry or the entry above is locked
if (!pmpcfg_q[index].locked && !(pmpcfg_q[index+1].locked && pmpcfg_q[index].addr_mode == riscv::TOR)) begin
```

Could you open a PR with this change?

omerguzelelectronicguy, 2023-08-14:
I made it but it is auto-blocked. Did I made something wrong?

Moschn, 2023-08-14:
Could you sign the eclipse agreement with the same email that you made the commit? Also, you might need to sign the commit with `git commit --amend -s`.

omerguzelelectronicguy, 2023-08-15:
Should I make a new pull request?
I sign the eclipse agreement but I was made PR from Web. To run this command `git commit --amend -s` I need to clone project right?

omerguzelelectronicguy, 2023-08-15:
I think it will take your time than necessary. Is it better to make this change by an active developer?

PR #1335 (2023-08-15): Fix index calculation for locked PMP csr logic
Fixes #1332
Superseeds #1334
commit 306973a3 (2023-08-15): Fix index calculation for locked PMP csr logic (#1335)

Fixes #1332
Superseeds #1334
closed 2023-08-15 by zarubaf