openhwfoundation/cva6 #1338: RVFI Limitation

AyoubJalali, 2023-08-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
this issue is about the RVFI tracer that drive all sort of information related to an instruction (GPR, PC, CSR , MEM_ADDR ...), the problem or the limitation in this tracer is that it can't drive information of an instruction does not been commited (killed or raised an exception), I really want to think about a possibility to update this tracer to be more practical in that way, because it affect ISACOV coverage.

JeanRochCoulon, 2023-09-12:
To my mind, an information is missing: getting an exception without the related instruction is a mess to monitor the cva6 behaviour. I confirm the need to fix it.

JeanRochCoulon, 2023-11-08:
Related to #1441 
RVFI will be revisited by @Gchauvon, I reassigned this issue.

MikeOpenHWGroup, 2023-12-08:
Please keep in mind that we want to use the [OpenHW version of RVFI](https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/rvfi.html), as specified in the CV32E40S User Manual.  It is a super-set of the SymbioticEDA version and contains information needed to operate the ISS as a reference model (e.g. Spike Tandem).

zchamski, 2024-08-23:
Now RVFI provides extensive information when exceptions are triggered.  @AyoubJalali : is the information provided by RVFI sufficient?  If so, can you close this issue?

AyoubJalali, 2024-08-23:
> Now RVFI provides extensive information when exceptions are triggered. @AyoubJalali : is the information provided by RVFI sufficient? If so, can you close this issue?

good for me

closed 2024-08-23 by AyoubJalali