openhwfoundation/cva6 #1342: Decoding LWU as a RV32 instruction

AyoubJalali, 2023-08-23:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
I try inject some RV64 instruction with RV32, just to be sure that CVA6 raise an exception on RV64 instruction because the STEP1 configuration is a RV32, so I notice that CVA6 is decoding **LWU** as RV32 instruction.
[image]
to reproduce the BUG you just need to inject a **LWU** (in binary format) for example try to add in your test.S   ->  **### .4byte 0x6f9a6f83**

JeanRochCoulon, 2023-08-28:
Hello @AyoubJalali 
The merged PR #1344 fix the issue, can you confirm it and close the issue ?

AyoubJalali, 2023-08-29:
Great You Merge the Fix 
Yes I confirm the PR #1344 fix the issue

AyoubJalali, 2023-08-29:
The bug is fixed

closed 2023-08-29 by AyoubJalali