openhwfoundation/cva6 #1345: PMU counter, L2 chache, CC ( cache coherence )

HimalSuebdi, 2023-08-25:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Is the PMU counter being implemented in the L2 cache? 

1. Does cva6 have a performance monitoring unit/counter (PMU/PMC)? If yes, can you please show me a direction in Git Hub? 

2. I also wonder how to track the traffic at the L1 cache level.  
3. Which kind of cache coherence protocol is implemented? &  How is it implemented? Can you please show me a direction in Git Hub?

fatimasaleem, 2023-08-28:
Hi @HimalSuebdi , 

1. Currently, CVA6 can only use the L2 Cache of OpenPiton's mesh, there is not any L2 Cache available in the CVA6 repo.
2. CVA6 is using write-through L1 data cache, which you can track by tracking [dcache_req_ports_i](dcache_req_ports_o ) & [dcache_req_ports_o](https://github.com/openhwgroup/cva6/blob/e1c40c5dcf89e2bf8b2d26349778d59742c013ed/core/cva6.sv#L921C6-L921C25)  in CVA6 top.
3. Since only L1 write through cache subsystem is present, there is not need to have any cache coherency protocol in CVA6. 
But you can find the coherency protocol in openpiton mesh, [here](https://parallel.princeton.edu/openpiton/docs/micro_arch.pdf) is a documentation link for it.

JeanRochCoulon, 2023-08-28:
@HimalSuebdi 
Are you satisfied by @fatimasaleem answer ?  In that case, can you close the issue
Thanks

closed 2023-08-28 by HimalSuebdi