openhwfoundation/cva6 #1347: Performance bug store buffer

YasasSeneviratne, 2023-08-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

### Underutilization of the speculative store buffer

From my understanding, the speculative store buffer should be ready to accept stores either when there is an empty buffer entry or at a commit.  In the current implementation, the store_buffer_ready signal is set in the following file.

core/store_buffer.sv:81 `ready_o = (speculative_status_cnt_q < (DEPTH_SPEC - 1)) || commit_i;`

When this implementation was simulated using Verilator, I found out that the above condition used for setting ready_o is not the optimum. Following is the trace for the simulation I did with two speculative sore buffer entries. The two `valid` waveforms are for the two speculative store buffer entries. (Note: ready_o is connected to store_buffer_ready)

[image]

In the clock cycle that is marked, both speculative store buffers are occupied. However, in the following cycle speculative buffer 0 is ready to accept stores. This means that the  store_buffer_ready should be high. However, because of the logic used it is still low resulting in a speculative store buffer underutilization. I think the store_buffer_ready signal should be set based on store_buffer entry validness which will better utilize resources.

JeanRochCoulon, 2023-08-28:
@cathales This issue is relative to performance. Maybe Have you notice the same limitation ?

fatimasaleem, 2023-08-29:
@JeanRochCoulon what if we do `ready_o = (speculative_status_cnt_q < (DEPTH_SPEC)) || commit_i;` instead of `ready_o = (speculative_status_cnt_q < (DEPTH_SPEC - 1)) || commit_i;`. I think this will resolve this limitation.

fatimasaleem, 2023-08-31:
Hi @YasasSeneviratne  can you confirm that [PR#1349](https://github.com/openhwgroup/cva6/pull/1349) resolves this for you as well? Thanks

closed 2023-10-07 by JeanRochCoulon