openhwfoundation/cva6 #1350: [BUG] Tracing Verilator 5

Gchauvon, 2023-08-29:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

PR https://github.com/openhwgroup/cva6/pull/1341 removed the tracing with Verilator5 when installed from [core-v-verif](https://github.com/openhwgroup/core-v-verif).
I believe VERILATOR_ROOT variable is changed for VERILATOR_INSTALL_DIR.
 
Would be great to update CVA6 [README.md ](README.md) to use Verilator5 withtout  core-v-verif because we have some users doing so. 

@zchamski can you look into this ?

JeanRochCoulon, 2023-10-24:
related to #1569

zchamski, 2023-12-12:
VL5 tracing issue linked to installation method (caused by #1341) was fixed in commit 5b37393a2e (see  PR #1395) so that VL tracing setup depends only on variable VERILATOR_INSTALL_DIR.

The remaining task is to update the latest README.md (heavily reworked in 6da3f465) to include a "Waveforms" section, possibly after section "Logs".

JeanRochCoulon, 2023-12-12:
If ok with thte status, @Gchauvon please close the issue

closed 2023-12-12 by Gchauvon