openhwfoundation/cva6 #1352: Decoding RV64 as a RV32 instruction

AyoubJalali, 2023-08-30:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I generate some illegal instruction (illegal opcode **_0x232532af_**), and I found that an instruction is actually a **RV64A**  instr, the problem is the **CV32A60X** decode it, without raising an exception.
So I think it's a BUG in the decoder, if you wanna reproduice the BUG you just need to inject the instruction 0x232532af in your test.S

JeanRochCoulon, 2023-08-31:
Do you have more information about the 0x232532af instruction: mnemonic for instance. Because I did not find it in Risc-V information (maybe not up-to-date).

AyoubJalali, 2023-08-31:
> Do you have more information about the 0x232532af instruction: mnemonic for instance. Because I did not find it in Risc-V information (maybe not up-to-date).

it's RV64A instruction (0x232532af) **amoxor.d t0, s2, (a0)**

PR #1355 (2023-09-01): Decode AMOXOR_D (and similar instructions) only when XLEN=64bits 
It fixes #1352
closed 2023-09-01 by JeanRochCoulon