openhwfoundation/cva6 #1353: [BUG] Debugging with OpenOCD v12.0

hasanalshaikh, 2023-08-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Running into the following error while trying to debug on Genesys 2 board through OpenOCD (v12.0). The error message is not really that helpful for me. I am not sure what the problem is. 

> Open On-Chip Debugger 0.12.0+dev-03014-gee5c5c292 (2023-08-09-19:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'ftdi layout_signal' not 'ftdi_layout_signal'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
riscv
  riscv authdata_read [index]
  riscv authdata_write [index] value
  riscv dm_read reg_address
  riscv dm_write reg_address value
  riscv dmi_read address
  riscv dmi_write address value
  riscv dump_sample_buf [base64]
  riscv etrigger set [vs] [vu] [m] [s] [u] <exception_codes>|clear
  riscv exec_progbuf instr1 [instr2 [... instr16]]
  riscv expose_csrs n0[-m0|=name0][,n1[-m1|=name1]]...
  riscv expose_custom n0[-m0|=name0][,n1[-m1|=name1]]...
  riscv hide_csrs {n0|n-m0}[,n1|n-m1]......
  riscv icount set [vs] [vu] [m] [s] [u] [pending] <count>|clear
  riscv info
  riscv itrigger set [vs] [vu] [nmi] [m] [s] [u] <mie_bits>|clear
  riscv memory_sample bucket address|clear [size=4]
  riscv repeat_read count address [size=4]
  riscv reset_delays [wait]
  riscv resume_order normal|reversed
  riscv set_bscan_tunnel_ir value
  riscv set_command_timeout_sec [sec]
  riscv set_ebreakm on|off
  riscv set_ebreaks on|off
  riscv set_ebreaku on|off
  riscv set_enable_virt2phys on|off
  riscv set_enable_virtual on|off
  riscv set_ir [idcode|dtmcs|dmi] value
  riscv set_maskisr ['off'|'steponly']
  riscv set_mem_access method1 [method2] [method3]
  riscv set_reset_timeout_sec [sec]
  riscv use_bscan_tunnel value [type]
riscv.cpu
  riscv.cpu arm
    riscv.cpu arm semihosting ['enable'|'disable']
    riscv.cpu arm semihosting_basedir [dir]
    riscv.cpu arm semihosting_cmdline arguments
    riscv.cpu arm semihosting_fileio ['enable'|'disable']
    riscv.cpu arm semihosting_read_user_param
    riscv.cpu arm semihosting_redirect (disable | tcp <port>
              ['debug'|'stdio'|'all'])
    riscv.cpu arm semihosting_resexit ['enable'|'disable']
  riscv.cpu arp_examine ['allow-defer']
  riscv.cpu arp_halt
  riscv.cpu arp_halt_gdb
  riscv.cpu arp_poll
  riscv.cpu arp_reset
  riscv.cpu arp_waitstate
  riscv.cpu array2mem arrayname bitwidth address count
  riscv.cpu cget target_attribute
  riscv.cpu configure [target_attribute ...]
  riscv.cpu curstate
  riscv.cpu eventlist
  riscv.cpu examine_deferred
  riscv.cpu get_reg list
  riscv.cpu invoke-event event_name
  riscv.cpu mdb address [count]
  riscv.cpu mdd address [count]
  riscv.cpu mdh address [count]
  riscv.cpu mdw address [count]
  riscv.cpu mem2array arrayname bitwidth address count
  riscv.cpu mwb address data [count]
  riscv.cpu mwd address data [count]
  riscv.cpu mwh address data [count]
  riscv.cpu mww address data [count]
  riscv.cpu read_memory address width count ['phys']
  riscv.cpu riscv
    riscv.cpu riscv authdata_read [index]
    riscv.cpu riscv authdata_write [index] value
    riscv.cpu riscv dm_read reg_address
    riscv.cpu riscv dm_write reg_address value
    riscv.cpu riscv dmi_read address
    riscv.cpu riscv dmi_write address value
    riscv.cpu riscv dump_sample_buf [base64]
    riscv.cpu riscv etrigger set [vs] [vu] [m] [s] [u]
              <exception_codes>|clear
    riscv.cpu riscv exec_progbuf instr1 [instr2 [... instr16]]
    riscv.cpu riscv expose_csrs n0[-m0|=name0][,n1[-m1|=name1]]...
    riscv.cpu riscv expose_custom n0[-m0|=name0][,n1[-m1|=name1]]...
    riscv.cpu riscv hide_csrs {n0|n-m0}[,n1|n-m1]......
    riscv.cpu riscv icount set [vs] [vu] [m] [s] [u] [pending]
              <count>|clear
    riscv.cpu riscv info
    riscv.cpu riscv itrigger set [vs] [vu] [nmi] [m] [s] [u]
              <mie_bits>|clear
    riscv.cpu riscv memory_sample bucket address|clear [size=4]
    riscv.cpu riscv repeat_read count address [size=4]
    riscv.cpu riscv reset_delays [wait]
    riscv.cpu riscv resume_order normal|reversed
    riscv.cpu riscv set_bscan_tunnel_ir value
    riscv.cpu riscv set_command_timeout_sec [sec]
    riscv.cpu riscv set_ebreakm on|off
    riscv.cpu riscv set_ebreaks on|off
    riscv.cpu riscv set_ebreaku on|off
    riscv.cpu riscv set_enable_virt2phys on|off
    riscv.cpu riscv set_enable_virtual on|off
    riscv.cpu riscv set_ir [idcode|dtmcs|dmi] value
    riscv.cpu riscv set_maskisr ['off'|'steponly']
    riscv.cpu riscv set_mem_access method1 [method2] [method3]
    riscv.cpu riscv set_reset_timeout_sec [sec]
    riscv.cpu riscv use_bscan_tunnel value [type]
  riscv.cpu set_reg dict
  riscv.cpu smp [on|off]
  riscv.cpu smp_gdb
  riscv.cpu was_examined
  riscv.cpu write_memory address width data ['phys']
ariane.cfg:36: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "ariane.cfg", line 36

hasanalshaikh, 2023-08-31:
Okay so, I got it running with openocd v0.10. It seems the config, however cva6 is set up, only supports that specific version. Is there any plan to provide compatibility with newer openocd versions? I personally had to jump through a whole host of loops to compile that 5 year old version of openocd.

closed 2023-08-31 by hasanalshaikh

namnguyen269, 2023-10-26:
@hasanalshaikh 
Congratulations, you have successfully fixed the error. It would be great if you could provide me with the steps to set up and use openocd for debugging.

hasanalshaikh, 2023-10-26:
> @hasanalshaikh Congratulations, you have successfully fixed the error. It would be great if you could provide me with the steps to set up and use openocd for debugging.

I assume that you have successfully managed to generate the bitstream and then uploaded it to your FPGA board. Now, you will need a compatible openocd. Use the [RISC-V fork of openocd](https://github.com/riscv/riscv-openocd). When I closed this issue, I checked out an older version of the repo and had to edit the source code to get it to compile properly. There is however an easier way. Build the latest version (v0.12) and you won't have to go through the trouble of modifying the source code of openocd. To build the latest version, just follow their github repo, install the prerequisites. Then do: 
```
sudo apt-get install libftdi-dev
sudo apt-get install libusb-1.0-0
sudo apt-get install libusb-1.0-0-dev
then clone openocd github
./bootstrap
./configure --enable-ftdi
make
sudo make install
```
You should now be able to see your device listed when you enter `lsusb` on your terminal. Next, you need to make a very simple modification in line 36 of the provided ariane.cfg file in the cva6 repo which is `riscv set_prefer_sba off` with the equivalent command available [here](https://sourceforge.net/p/openocd/mailman/message/37389126/). 

Once you do that, execute `openocd -f ariane.cfg` in a terminal which should give you the "Listening to gdb connections on port 3333". Keep this terminal on and on another terminal open gdb and follow the instructions detailed in this repo. Hope this helps!