openhwfoundation/cva6 #1354: file lose for Ariane 4.2

hhhsiang, 2023-08-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,

I am currently engaged in academic research on CPU architecture, and I have chosen CVA6 as my baseline. As part of my work, I intend to modify certain blocks within the CVA6 design. Consequently, I need to generate bitstream for deployment on the Genesys-2 FPGA board.

I have identified the [Ariane 4.2](https://github.com/openhwgroup/cva6/releases/tag/v4.2.0) repository as a stable version for my purpose. To proceed, I downloaded the corresponding source code ZIP file onto my PC. For simulation purposes, I am using Vivado 2018.2. However, during my simulation process, I encountered an issue where certain files, such as 'dm', appear to be missing. This led to an error indicating that 'dm' was not declared. Interestingly, I found the 'dm' file on GitHub at the following location: [dm_pkg.sv in riscv-dbg](https://github.com/pulp-platform/riscv-dbg/blob/6d768ac6371356b6e382c01ae7d860dd3e45248b/src/dm_pkg.sv).

After checking the downloading source code file on my PC, I noticed that the 'riscv-dbg/src/' directory was empty, and there was no 'dm_pkg' file present.

This is ariane_pkg.sv(2019version), and it needs dm_pkg.
[image]
 It references the 'dm_pkg', due to its absence, I am unable to proceed.

My question is: would it be acceptable to copy the 'dm_pkg' file directly from GitHub into my local repository? Is this approach likely to introduce unforeseen errors? Alternatively, should I consider abandoning the use of the Ariane 4.2 version (2019) and explore the current repository instead?

Thank you.

PeterRugg, 2023-09-12:
I suspect the issue is you need to run `git submodule update --init --recursive` inside the CVA6 repository: that should populate the empty directories, which are submodules. See, e.g. https://www.atlassian.com/git/tutorials/git-submodule.

hhhsiang, 2023-09-13:
@PeterRugg , thank you for your assistance! However, I've encountered another issue. I downloaded the 2019 version from this link: https://github.com/openhwgroup/cva6/releases/tag/v4.2.0.
[image]

Since I downloaded it on **Windows**, I'm not sure where to execute the following command:
`git submodule update --init --recursive`

Once again, I appreciate your help!

PeterRugg, 2023-09-13:
Okay, I doubt there's a nice way to get this working without installing git. I believe there are options to install git on Windows.
If that's really not an option, then I think the best solution might be to browse through the v4.2.0 tag of the repository on the web interface, e.g. https://github.com/openhwgroup/cva6/tree/v4.2.0/src, and see where there are submodules (shown for me with an "@" and then the commit of the submodule referred to). You can then follow each of the links to the corresponding submodules and download the files from that version as a zip, and paste them into the corresponding directory. Sounds very messy though!
FYI: I'm not related to CVA6, was just browsing and thought I might be able to help.

hhhsiang, 2023-09-15:
Ok, I got it. Thank you so much ! I will try it.

zarubaf, 2023-10-23:
This has been answered! Thanks @PeterRugg!

closed 2023-10-23 by zarubaf