openhwfoundation/cva6 #1363: Question on CVA6 PMPCFG CSR legal values

spidugu444, 2023-09-06:
Hi,

As per the riscv privileged specification, for PMPCFG CSR bit [6:5] marked as WARL, may I know what the legal value for such fields in CVA6?

[image]

Note: **When I wrote non-zero values in bit [6:5] during PMPCFG CSR access mode verification, I was reading same non-zero values.**

Thanks

Moschn, 2023-09-06:
The section you cite in the spec says that bits 7, 4, 3, 2, 1, 0 are valid configuration values. Bits 5 and 6 are reserved. 

CVA6 follows this convention. All bits are writeable and will be stored. Bits 5 and 6 will never be used by CVA6 but they are stored and will be returned as written.

Note that not all PMPCFG CSRs may be enabled based on the configuration of the core. In the default configuration, only 8 PMP regions are enabled, and therefore, writes to PMPCFG[2,3] have no effect.

spidugu444, 2023-09-07:
Hi @Moschn, Thanks for your quick response.

spidugu444, 2023-09-21:
Hi @moschin,

while verifying access modes for PMPCFG CSRs  I had below queries could you please help me on this.

	1. As per riscv priviliged specification it is mentioned 
        When G ≥ 1, the NA4 mode is not selectable. As in CVA6 as we had G=1 what will be the expected behaviour in CVA6 when we select Address as NA4.

2. In riscv priviliged specification it is mentioned as The R, W, and X fields form a collective WARL field for which the combinations with R=0 and W=1 are reserved.
What will be the expected behaviour in CVA6 when we make R=0 and W=1.

While verifying access modes for PMPCFG CSR when I tried combination R=0 and W=1, I am reading PMPCFG CSRs previous value is this behavior expected in CVA6 ? on spike end I am reading different value for this combination(R=0 and W=1).

Thanks

Moschn, 2023-09-21:
> 1. As per riscv priviliged specification it is mentioned
>     When G ≥ 1, the NA4 mode is not selectable. As in CVA6 as we had G=1 what will be the expected behaviour in CVA6 when we select Address [...]

The current behavior of CVA6 is that it checks if any PMPCFG (or PMPADDR) is legal or not. If it is illegal it will ignore the write and continue with the old value. This applies to PMPCFGs with A set to NA4 or R=0 and W=1. This behavior follows the spec as the PMP CSRs are WARL (write any read legal).

I do not know what Spike does but the RISC-V spec just states that any core has to always read legal values. It is plausible that spike does something different than CVA6 in these cases.

spidugu444, 2023-10-09:
@Moschn, Thanks for your reply.

closed 2023-10-09 by spidugu444