openhwfoundation/cva6 #1364: Integrate the debug_test into the CVA6 environment.

MikeOpenHWGroup, 2023-09-06:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

.

### Description of Done

.

### Associated PRs

.

JeanRochCoulon, 2023-09-11:
The embedded configuration does not support debug mode. that's why this task should be dedicated to step1 configuration, or another configuration using debug feature. When label indicating the configuration will be created, we will be able to label the task.

JeanRochCoulon, 2024-05-22:
@xiaoweish You shared the idea to work on debug. Do you confirm?   In that case, is this task describe what should be done ?

xiaoweish, 2024-06-12:
Hi @JeanRochCoulon,
Sorry for missing your comment!
Yes, it’s on my task list. I’ll update it once ready!
Thank you!

Xiaowei

xiaoweish, 2024-06-17:
Hi @MikeOpenHWGroup 

As to step 1 in `How to?`

> 1. Add a [debug control virtual peripheral](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/test_programs.html#virtual-peripherals) to the testbench and connect it to the [debug_req_i](https://github.com/openhwgroup/core-v-verif/blob/8dac926e4c7693c52609572b4f8281eec30c9823/cva6/tb/uvmt/cva6_tb_wrapper.sv#L78) input of the core.

In the webpage, it says,
> The use of the interrupt timer control and instruction memory stall controller are not well understood and it is possible that none of the testscases inherited from the RISC-V foundation or the PULP-Platform team use them. As such they are likely to be deprecated and their use by new test programs developed for CORE-V is strongly discouraged.

So, to verify interrupt and debug_req, would you please share your thoughts here?
1. Does above 'deprecated' statement apply to debug test also?
2. Use the directed-test-program called the [debug_test](https://github.com/orgs/openhwgroup/projects/3/views/cv32e40p/tests/programs/custom/debug_test) or UVM based debug cases? which would be better for now? Of course it might involve `tandem verification` if using UVM. 

Thanks Mike!