openhwfoundation/cva6 #1372: [BUG] rvfi_tracer.sv: No valid address of `tohost` - Critical warning?

pcotret, 2023-09-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

```
[rvf_tracer] WARNING: No valid address of 'tohost' (tohost == 0x00000000000000), termination possible only by timeout or Ctrl-C!
```
I was playing with the [PMP riscv-test](https://github.com/riscv-software-src/riscv-tests/tree/master/benchmarks/pmp) and I got the warning above. The execution on a default CVA6 ended  with a `SUCCESS` which is great.

However, I still don't clearly understand if I should worry about this warning.

zarubaf, 2023-09-08:
The tracer expects a `.tohost` symbol in the elf for host communication (especially exit). If that symbol isn't there (because the linker didn't put it there) the test only exits on timeout or ctrl-c.

@JeanRochCoulon for sure knows more on the exits status.

zchamski, 2023-10-24:
Last year the decision was taken to use HTIF-style termination (store of a specific value at address `tohost`) as a universal end-of-test marker for CVA6 testing.

The reasons for this choice were twofold:
* compatibility with Spike
* reliance on a memory write at a symbolic address instead of a specific instruction (ECALL) to enable the verification of all instructions including ECALL and all memory locations.

In the official HTIF end-of-test approach, a write of an *odd* 64-bit value at symbolic address `tohost` is treated as an end-of-test signal, with bits 47:1 interpreted as the test exit code and bits 63:48 expected to be all zeroes.  In CVA6 RVFI-based end-of-test approach, this condition was simplified to "an XLEN bits wide write of an odd value at symbolic address `tohost`".  Hence, the knowledge of the value of `tohost` is needed to enable proper termination detection *in this approach*.

The value can either be extracted from the ELF file, or supplied on the RTL simulator command line via the plus-arg `tohost_addr`.

Obviously, other termination mechanisms are possible, so the `WARNING` appearing in the message could be demoted to `NOTICE` level.

JeanRochCoulon, 2023-11-08:
@valentinThomazic