openhwfoundation/cva6 #1375: Read, Write & Execution on 0x0 address

AyoubJalali, 2023-09-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, 
So I notice that the CVA6 has permission to read, wrote & execute on 0x0 address, I think it should raise an exception on lower address.
to reproduce the Bug you'll find an .S

[bug_test.S.txt](https://github.com/openhwgroup/cva6/files/12560778/bug_test.S.txt)

AyoubJalali, 2023-12-19:
what I observed is the CVA6 can read/ write and execute from address 0x0 to 0xffff_ffff (virtual address), we need to resolve this, this is not right

JeanRochCoulon, 2025-01-24:
As CV32A65X does not have PMA checkers, it can read, write and fetch from/to all addresses. I remove the cv32a65x.