openhwfoundation/cva6 #1377: CVA6 RTL Lint warnings for CV32A60X

fatimasaleem, 2023-09-09:
## Description

RTL Linting is used to verify adherence to specified design coding rules that prevent synthesis issues, functional bugs and enforce coding styles for readability & re-use. For that purpose, we have to enable lint warnings and remove/waive them where necessary.

## Changes

Track and resolve the lint warnings in `core/` directory

## Current Status

This table specifies all the warnings related to lint in CVA6 RTL and classifies those into specific categories. The progress of all the Lint Issues will be maintained here.
  	 	 	 	
| Type of Lint Warning       | No. of Warnings | Status | PR |
|------------|-------|--------|-------|
| MODDUP     | 1     | ToDo   | None  |
| SELRANGE   |      | ToDo   | None  |
| WIDTHEXPAND|      | In Progress   | None  |
| WIDTHCONCAT|      | In Progress   | None  |
| WIDTHTRUNC |      | In Progress   | None  |
| UNOPTFLAT  |      | ToDo   | None  |

```

```

## Description of Done

Resolution of cva6/core/ directory lint problems
All the related lint waivers are in place

JeanRochCoulon, 2023-10-10:
#1510 and #1520 have reduced the number of warnings. Do you have an udpdate of the above table to present this afternoon (West Shifted Meeting)?

fatimasaleem, 2023-10-19:
After the merge of the current PR, the status will be as follows

Type of Lint Warning | No. of Warnings | Status | PR
-- | -- | -- | --
MODDUP | 0 | Done | None
SELRANGE | 13 | In Progress | None
WIDTHEXPAND | 34 | In Progress | None
WIDTHCONCAT | 1 | In Progress | None
WIDTHTRUNC | 7 | In Progress | None
UNOPTFLAT | 12 | In Progress | None

fatimasaleem, 2023-10-24:
Type of Lint Warning | No. of Warnings | Status
-- | -- | -- 
MODDUP | 0 | Done 
SELRANGE | 11 | P2 
WIDTHEXPAND | 2 | P2 
WIDTHCONCAT | 0 | Done 
WIDTHTRUNC | 0 | Done

JeanRochCoulon, 2024-05-22:
@fatimasaleem  Can you consider this task over ?   The current Spyglass task will help to fix LINT errors.

closed 2024-05-22 by fatimasaleem