openhwfoundation/cva6 #1381: [BUG] RVFI `rvfi_valid`

zarubaf, 2023-09-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

For the [RVFI](https://github.com/SymbioticEDA/riscv-formal/blob/master/docs/rvfi.md) my assumption was that the `rvfi_valid` should go high when the core retires an instruction. I thought this should also hold true for exceptions (which would count as retired instructions, but with an exception annotated).

For the current implementation this isn't the case for CSR instructions that cause an exception: https://github.com/openhwgroup/cva6/blob/f7b79535dfc6115144585cefb973894add44a85c/core/commit_stage.sv#L159

My question is this intended or necessary (because for example spike comparison requires it)? Or a bug? If the latter I could propose a fix.

/cc @michael-platzer

AyoubJalali, 2023-09-12:
Hello @zarubaf, is it the same as this issue ? https://github.com/openhwgroup/cva6/issues/1338

JeanRochCoulon, 2023-09-12:
@MikeOpenHWGroup as you maybe have more visibility on what has been done for embedded cores, I assigned this issue to you

zarubaf, 2023-09-12:
Oh yeah exactly that issue. I should have taken 

> I have searched the existing bug issues

more seriously. Sorry about that.

closed 2023-09-12 by zarubaf