openhwfoundation/cva6 #1385: Dual issue

JeanRochCoulon, 2023-09-12:


JeanRochCoulon, 2023-10-03:
Hi Jay
By the end of the year, the dual issue will be functional. Then verification could start..
Cheers

jay1sung, 2023-10-03:
You said "The progress indicator is the step done (functional single- and dual-issue, no impact on single-issue and expected performance on dual-issue)." So I understood that you already completed the function/performance check. You mentioned a model that is software language not RTL? 

Thanks,

Jay

JeanRochCoulon, 2023-10-04:
That is a python model which (tries to) estimate the performance gain.
Regards

jin8495, 2024-04-01:
Hi, @JeanRochCoulon.
I am interested in the dual issue feature. But this Github issue has had no progress since October.
I'm just wondering how things are going.

Thank you.

jin8495, 2024-04-03:
@cathales Thank you for the information. That is really what I want. 😄

cathales, 2024-04-05:
Update: ETA is July the 5th

Cc @jin8495

cathales, 2024-04-16:
Notes from https://github.com/openhwgroup/cva6/pull/2047 for parametrization of superscalar (feel free to discuss these points)

- Need two options `bit SuperscalarEn` and `int unsigned PIP_WIDTH`
- `NrCommitPorts` can be replaced by `PIP_WIDTH`
- Try to make Verilator understand that `[1]` is never used unless `SuperscalarEn`
- `core/frontend/instr_queue.sv` Update `NID` and its comment

cathales, 2025-01-06:
I think all follow-ups have been completed.

closed 2025-01-06 by cathales