openhwfoundation/cva6 #1387: Setup and maintain embedded CI on Thales servers

JeanRochCoulon, 2023-09-12:
***What*** Provide code coverage metrics in an automated way without consuming too much energy.

***Howto*** First do not run heavy tests if smoke tests fail. Then add another workflow running on `master` once a week, to provide metrics and their history.

***Current status*** There are currently no metrics available.

***Risks*** The amount of code and infrastructure to change might be significant.

***Prerequisites*** Complete CI simplification before adding new things.

***KPI*** None

***Description of done*** The metrics dashboard is available.

JeanRochCoulon, 2023-10-10:
Would be useful to refer the related PR.

cathales, 2023-10-19:
This description includes notion of code coverage: https://github.com/openhwgroup/cva6/issues/1436

However this current task focuses on "when is each job run?".

cathales, 2023-10-19:
Notes from meeting:

- #1436: when we have coverage results on several weeks, add a graph with coverage result in the dashboard
- Add a new color (gray?) for ignored tests: they are run for information but they are allowed to fail without making the whole pipeline fail. (Needs to update `report_generator.py` for report generation and the dashboard for report rendering)

cathales, 2024-01-26:
We wanted to ignore tests that are known to be failing on master. As we do not run these tests before merging PRs, do we still want to ignore them?

Note: the failing tests are in the `compliance` and `riscv-tests-p` jobs. Actually these tests are "crashing", not only "failing". This implies that subsequent tests in the same job are not run. So we thought that 2 tests were failing (one in each job) while more are crashing! I had to list them to debug my own stuff so I share it with you below.

List of crashing tests on c430c6c34be0e24175e03d3c91478e349bfaaf4e (2024-01-05):

- In `compliance` (`verif/tests/testlist_riscv-compliance-cv32a6_embedded.yaml`)
    - `rv32uc-rvc`
    - `rv32ui-jal`
    - `rv32ui-bge`
    - `rv32ui-blt`
    - `rv32ui-bgeu`
    - `rv32ui-sw`
    - `rv32ui-lbu`
    - `rv32ui-sb`
    - `rv32ui-slti` Note: this test, compiling `slti.S`, is actually named `rv32ui-sw` in the test list. Looks like a typo.
    - `rv32ui-sra`
    - `rv32ui-srl`
    - `rv32ui-sh`
    - `rv32ui-lw`
    - `rv32ui-andi`
    - `rv32ui-srli`
    - `rv32ui-slli`
    - `rv32ui-beq`
    - `rv32ui-sll`
    - `rv32ui-addi`
    - `rv32ui-lh`
    - `rv32ui-and`
    - `rv32ui-xori`
    - `rv32ui-sub`
    - `rv32ui-slt`
    - `rv32ui-lb`
    - `rv32ui-or`
    - `rv32ui-lui`
    - `rv32ui-ori`
    - `rv32ui-bltu`
    - `rv32ui-fence_i`
    - `rv32ui-auipc`
    - `rv32ui-srai`
    - `rv32ui-jalr`
    - `rv32ui-xor`
    - `rv32ui-simple`
    - `rv32ui-lhu`
    - `rv32ui-bne`
    - `rv32ui-add`
    - `rv32ui-sltiu`
    - `rv32ui-sltu`
    - `rv32si-sbreak`
    - `rv32si-scall`
    - `rv32si-ma_fetch`
    - `rv32si-wfi`
    - `rv32si-csr`
    - Note that there are 2 failing tests, `rv32mi-scall` and `rv32mi-csr`, but at least the job runs to completion after removing the above tests.
- In`riscv-tests-p` (`verif/tests/testlist_riscv-tests-cv32a6_embedded-p.yaml`)
    - `rv32mi-p-scall`
    - `rv32si-p-csr`
    - `rv32si-p-ma_fetch`
    - `rv32si-p-scall`
    - `rv32si-p-wfi`
    - `rv32si-p-sbreak`
    - Note that there is 1 failing test, `rv32mi-p-csr`, but at least the job runs to completion after removing the above tests.

The crashes seem related to the two letters after `rv32`.

cathales, 2024-03-18:
Can we consider this task as done?

cathales, 2024-03-19:
Done #1944

closed 2024-03-19 by cathales