openhwfoundation/cva6 #1398: Add coverage dedicated to Zicond, Zb and Zce to ISACOV

JeanRochCoulon, 2023-09-13:


JeanRochCoulon, 2023-10-10:
Any news ?   If some PRs have been merged, please refer the PR number.

AyoubJalali, 2023-10-10:
I add the coverage for Zicond (# [1473](https://github.com/openhwgroup/cva6/pull/1473))
For Zce is pending (do you mean Zcb ?)

JeanRochCoulon, 2023-10-17:
Zicond has been disabled in embedded configuration #1534

AyoubJalali, 2023-10-26:
Waiting for [the merge](https://github.com/openhwgroup/core-v-verif/pull/2257) of the ISA_DECODER into master, to use it to decode these new instruction, because the dasm spike does not support them.

JeanRochCoulon, 2024-05-22:
HEllo @AyoubJalali    What is the status ?    Can we consider it as completed ?

AyoubJalali, 2024-05-22:
For now there's only coverage for Zcb !!

AyoubJalali, 2024-11-05:
For now the CV32A65X does not support ZCMP or ZCMT, should I add covergoup dedicated for them ? @JeanRochCoulon