openhwfoundation/cva6 #1399: Implement new isntructions and generate random test with Zicond, Zcb, Zcmp in riscv-dv

JeanRochCoulon, 2023-09-13:


AyoubJalali, 2023-10-10:
I add [Zicond](https://github.com/openhwgroup/cva6/pull/1473) & [Zcb](https://github.com/openhwgroup/cva6/pull/1496) instructions into the generator cva6-dv 
for Zcmp I'm waiting for a spec

jquevremont, 2023-10-10:
@fatimasaleem is working on the spec. Zcmp expected e/o October.

JeanRochCoulon, 2023-10-17:
Zicond has been disabled in embedded configuration #1534

AyoubJalali, 2023-10-26:
A MR is ready to be approved

AyoubJalali, 2023-11-09:
Now we have zcb instructions in the core-dv, but still Zcmp (need a spec)

JeanRochCoulon, 2023-11-09:
Zcmp can be compiled by GCC master, and simulate with Spike.

AyoubJalali, 2023-11-23:
Zcmp is implemented due to the PR #1648

closed 2023-11-23 by AyoubJalali

JeanRochCoulon, 2023-11-30:
@ASintzoff  Can you review the task ?   If you think the task is completed, can you update the status from "Done" to "finished" ?