openhwfoundation/cva6 #1402: [TASK] Add CRSs to CVA6 RVFI

zchamski, 2023-09-13:


zchamski, 2023-09-14:
This task supersedes #1362.

As @MikeOpenHWGroup  commented in #1362,

> I recommend that we implement the RVFI tracer to the same specification as is used by the [CV32E40S core](https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/rvfi.html#risc-v-formal-interface). This is an extension of the original RVFI-specification from [SymbioticEDA](https://github.com/SymbioticEDA/riscv-formal/blob/master/docs/rvfi.md).
> 
> Furthermore, we should also consider implementing the CVA6 RVFI as _behavioral_ code that will not be synthesized into gates. This will save a very large number of I/O on the top-level of the core. A good example of how this is done is in the [cv32e40s wrapper](https://github.com/openhwgroup/cv32e40s/blob/5af3313008ec67965659c8788716e82f59bea21c/bhv/cv32e40s_wrapper.sv#L588).

MikeOpenHWGroup, 2023-11-21:
It is also worth considering the [RVVI-Trace](https://github.com/riscv-verification/RVVI/tree/main/RVVI-TRACE), an extension of the RVFI proposed by Imperas.  The best thing about RVVI is that it provides a nice set of timing diagrams to define the interface's operation in specific corner cases.

closed 2024-05-22 by JeanRochCoulon