openhwfoundation/cva6 #1403: [TASK] Modify Spike to match embedded configuration

zchamski, 2023-09-13:


MarioOpenHWGroup, 2023-10-19:
Further exploration is needed to catch all the discrepancies between spike and cva6

zchamski, 2023-10-19:
We have three sources of discrepancies:
* differences in interpretation of implementation-dependent "may"/"can" cases in the RISC-V specs
* bugs in CVA6
* (unlikely but possible) bugs in Spike.

We need to capture formally the "may" and "can" cases in RISC-V specs.  From there we need a systematic review of how these are implemented on both sides.

MarioOpenHWGroup, 2023-10-26:
tval value is not matching between core and spike, this needs to be discussed or fixed in spike side

MarioOpenHWGroup, 2023-11-09:
Tval has been reverted to the original state and the instr field of the scoreboard_entry_t has been removed

MarioOpenHWGroup, 2023-11-16:
Issues are appearing regarding specific cva6 config

MarioOpenHWGroup, 2023-11-23:
It is mandatory in this task that we know what we have the specification to rely when changing the Reference Model. Many issues above exist because unspecified parts or inconsistencies between the spec and the current environment implementation.

MarioOpenHWGroup, 2023-11-30:
I am making some notes about the can/may parts of the spec to implement (some of them) in Spike.

MarioOpenHWGroup, 2023-12-07:
No updates done in this task.

zchamski, 2023-12-12:
The behavior of bit 0 in the `PMPADDRn` CSRs depends on the address matching mode in effect (encoded in the associated `PMPCFG` register) **and** the PMP granularity set for each core.

The corresponding logic appears to be present in Spike and the termination of the PMP granularity is affected by the Device Tree specification used by Spike:

- A default value of 4 is set in the constructor of the `processor_t` class.
- If a device tree seen by Spike contain an explicit PMP granularity entry for the given core, that value will be used (if valid).

By default, Spike reads the built-in device tree which specifies a hard-coded PMP granularity of 4 (a compile-time constant).

On CVA6 the PMP granularity is hardwired to 8, and therefore any device tree-suppllied values will be silently ignored.  The best solution seems thus to be a new `Param` entry that overrides the default PMP granularity set by Spike.

zchamski, 2024-04-17:
> Agree, I will add the parameter

@MarioOpenHWGroup, I can see Spike params `/top/.../pmpregions` (number of PMP regions), `/top/.../pmpaddr0` (reset value of `PMPADDR0`), and `/top/.../pmpcfg0` (reset value of `PMPCFG0`), but none that can directly set PMP granularity.  Do you (plan to) have a specific parameter for this purpose?