openhwfoundation/cva6 #1404: [TASK] Compare Spike and RTL in lockstep

zchamski, 2023-09-13:


MarioOpenHWGroup, 2023-10-10:
A working version of the Tandem has been pushed to my repo that implements the following:
- Tandem comparing exceptions, interrupts, destination GPR, instruction and pc. 
- Common package that implements the functionality for the testharness and UVM implementation

MarioOpenHWGroup, 2023-10-19:
Tandem rebased to last cva6/master commit. Some discussion needed regarding integration Sim*.cc  `corev_apu/tb/common/Sim`  into riscv-isa-sim

MarioOpenHWGroup, 2023-10-26:
Fixing Github CI in order to merge.
Also fixing the issues commented in both PRs.

MarioOpenHWGroup, 2023-11-09:
CI has been fixed and it shows green on Thales CI and on Github

MarioOpenHWGroup, 2023-11-16:
PR has been merged already

MarioOpenHWGroup, 2023-11-23:
On going efforts to make the tandem available on verilator

MarioOpenHWGroup, 2023-12-07:
I am working on make verilator work on 64bits but it has issues in the DPI interface

zchamski, 2024-09-19:
Task completed for CV32A65X.  Adding other configurations is a matter of providing the necessary configuration files for Spike, and possibly, of adding futher configurability if needed.

closed 2024-09-19 by zchamski