openhwfoundation/cva6 #1407: [TASK] Condition RTL with A extension parameter

Gchauvon, 2023-09-13:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Adding `if(CVA6Cfg.*)` in RTL to identify the dead code and remove the dead gates from netlist. 
Based on if() directives, VCS is able to identify the unused code, this will allow to reach 100% code coverage. 

### Required Changes

Modification of multiple RTL files to remove gates if A extension is disabled

### Current Status

Not started

### Description of Done

Removal of all gates related to A extension in cva6 RTL when the A extension is disabled.

JeanRochCoulon, 2023-10-10:
@AEzzejjari  Have you started this task ?   If it is the case, do not hesitate to ask for the assignment and status changes (or do it if you have the rights)

JeanRochCoulon, 2023-12-05:
@AEzzejjari  Do you confirm this task is done ?   Can you link with the related PR

AEzzejjari, 2023-12-05:
The task has been completed, and the related PR number is 1554.

JeanRochCoulon, 2023-12-05:
I confirm. The modification has been merged, kgate have been reduced. To my mind, the task is finished.

closed 2023-12-05 by JeanRochCoulon