openhwfoundation/cva6 #1408: [TASK] Parameterize Privilege S/U

Gchauvon, 2023-09-13:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Parameterize Privileges S/U with `CVA6Cfg` parameter. Gates related to privileges S/U must be remove when not enable.
Privileges mode available will be : 

- Only M mode
- M/S/U mode

### Required Changes

RTL changes to condition gates related to privileges S/U.

### Current Status

Not started

### Description of Done

All gates related to S/U privileges can be disabled. CVA6 functionalities in M mode must be the same.

JeanRochCoulon, 2023-11-30:
This task has been done by Alae. @AEzzejjari can you link this task with the related PR please.

JeanRochCoulon, 2023-12-01:
Two new parameters have been added to the RTL to condition the S and U modes. RTL code has been gated by them. CC has been improved and CSR linked to U or S modes are removed from RTL.
LGTM. The task is finished.