openhwfoundation/cva6 #1409: [TASK] Select a configuration for caches in embedded configuration

Gchauvon, 2023-09-13:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Select the configuration for caches in embedded configuration (size, ways)

### Required Changes

Add the configuration to `cv32a6_embedded_config_pkg.sv`

### Current Status

In progress

### Description of Done

Configuration is written to the `cv32a6_embedded_config_pkg.sv`

JeanRochCoulon, 2024-05-22:
THis task should be completed, do you confirm @Gchauvon ?

Gchauvon, 2024-05-22:
No configuration has an embedded cache configuration yet.