openhwfoundation/cva6 #1410: [BUG] <title>

wang-hanmo, 2023-09-14:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello! I'm trying to setup CVA6 environment and I cannot run the script correctly. I followed the README and installed riscv-gnu-toolchain, then I ran `./verif/regress/smoke-tests.sh` but failed. Here's the error message:
```
Thu, 14 Sep 2023 12:58:51 INFO     Execution numero : 1
Thu, 14 Sep 2023 12:58:51 INFO     Compiling c test : ../tests/custom/hello_world/hello_world.c
Thu, 14 Sep 2023 12:58:51 INFO     /opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/libgcc.a(div.o)
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/../../../../riscv64-unknown-elf/bin/ld: warning: /home/whm/cva6/verif/sim/out_2023-09-14/directed_c_tests/hello_world.o has a LOAD segment with RWX permissions
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccrlpvgb.o: in function `verifyDouble':
/home/whm/cva6/verif/sim/../tests/custom/common/syscalls.c:36: undefined reference to `__eqdf2'
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/../../../../riscv64-unknown-elf/bin/ld: /home/whm/cva6/verif/sim/../tests/custom/common/syscalls.c:36: undefined reference to `__eqdf2'
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.3.1/../../../../riscv64-unknown-elf/bin/ld: /home/whm/cva6/verif/sim/../tests/custom/common/syscalls.c:39: undefined reference to `__nedf2'
collect2: error: ld returned 1 exit status

Thu, 14 Sep 2023 12:58:51 ERROR    ERROR return code: True/1, cmd:/opt/riscv/bin/riscv64-unknown-elf-gcc -mcmodel=medany -nostdlib          -nostartfiles ../tests/custom/hello_world/hello_world.c          -I/home/whm/cva6/verif/sim/dv/user_extension           -T../tests/custom/common/test.ld -g ../tests/custom/common/syscalls.c ../tests/custom/common/crt.S -lgcc -I../tests/custom/env -I../tests/custom/common -o /home/whm/cva6/verif/sim/out_2023-09-14/directed_c_tests/hello_world.o  -march=rv32imac_zicsr_zifencei -mabi=ilp32
```
I also tried to run the command `make verilate`, and it seems that my g++ don't support C++17, but my g++ version is g++-11. 
```
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/include -I/opt/riscv/include -I/opt/riscv/include -std=c++17 -I../corev_apu/tb/dpi -O3 -DVL_DEBUG  -DVL_THREADED -std=gnu++14 -Os -c -o ariane_tb.o ../corev_apu/tb/ariane_tb.cpp
In file included from /opt/riscv/include/fesvr/memif.h:10,
                 from /opt/riscv/include/fesvr/htif.h:6,
                 from /opt/riscv/include/fesvr/dtm.h:4,
                 from ../corev_apu/tb/ariane_tb.cpp:42:
/opt/riscv/include/fesvr/../riscv/cfg.h:99:8: error: ‘optional’ in namespace ‘std’ does not name a template type
   99 |   std::optional<reg_t>               start_pc;
      |        ^~~~~~~~
/opt/riscv/include/fesvr/../riscv/cfg.h:99:3: note: ‘std::optional’ is only available from C++17 onwards
   99 |   std::optional<reg_t>               start_pc;
      |   ^~~
``` 
Here is my riscv-gnu-toolchain:
```
whm@whm-virtual-machine:~/cva6$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/bin/../libexec/gcc/riscv64-unknown-elf/12.3.1/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/whm/riscv-gnu-toolchain/build/../gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv64-newlib --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gacfbdb1c271 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv64-newlib/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../../gcc --enable-multilib --with-multilib-generator='rv32e-ilp32e--;rv32i-ilp32--;rv32im-ilp32--;rv32iac-ilp32--;rv32imac-ilp32--;rv32imafc-ilp32f--;rv32imafdc-ilp32d--;rv64i-lp64--;rv64ic-lp64--;rv64iac-lp64--;rv64imac-lp64--;rv64imafdc-lp64d--;rv64im-lp64--;' --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.3.1 20230909 (gacfbdb1c271) 
```
Here is my gcc/g++:
```
whm@whm-virtual-machine:~/cva6$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-2ubuntu1~20.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-PfdVzN/gcc-11-11.4.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-PfdVzN/gcc-11-11.4.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.4.0 (Ubuntu 11.4.0-2ubuntu1~20.04) 
```
How could I fix the error?

hhhsiang, 2023-09-28:
hi, @wang-hanmo. I meet the same problem. Have you solved it?

pcotret, 2023-10-19:
Same issue here on a fresh machine (I know we shouldn't use `make verilate` anymore, but I'm on an older commit)

JeanRochCoulon, 2023-10-24:
Related to #1569

wang-hanmo, 2023-11-24:
> 👋 Hi there!
> 
> This issue seems inactive. Need more help? Feel free to update us. If there are no updates within the next few days, we'll go ahead and close this issue. 😊

OK. I tried many versions of Verilator and somehow fixed this problem. Thanks anyway!

closed 2023-11-24 by wang-hanmo