openhwfoundation/cva6 #1411: [TASK] Add Zc* in user guide and Requirement Specification

fatimasaleem, 2023-09-14:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

When Zc RTL code is merged, update the requirement specification and user guide to mention Zc* extension support

### Required Changes

Update Requirement spec & user manual

### Current Status

Zc is not mentioned in the specs

### Description of Done

merged PR

JeanRochCoulon, 2023-10-10:
Any news about this task ?  Needed by #1397

jquevremont, 2023-10-10:
Expected October 11th or 12th.

zarubaf, 2023-10-11:
@jquevremont I have added a `due` field to the project so that we can search/track 😄

jquevremont, 2023-10-11:
> @jquevremont I have added a `due` field to the project so that we can search/track 😄

@zarubaf where did you add it?

zarubaf, 2023-10-11:
[image]

Here, and its now a field in the projects list.

fatimasaleem, 2023-10-19:
Zcb is done. Zcmp is remaining.

PR #1536 (2023-10-19): [DOCS] Add Zcb Instructions in CVA6 user guide and requirement specification
This PR (partially) resolves #1411
closed 2023-10-19 by JeanRochCoulon