openhwfoundation/cva6 #1413: cebreak tests are failing in latest cva6 riscv-arch-test suite

spidugu444, 2023-09-14:
Hi All,

I have an issue with  cebreak tests while running in cva6 core for riscv-arch-test suite, it has some gpr mismatch between spike and verilator as shown below.
[cebreak.S](https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv32i_m/C/src/cebreak-01.S)

Please find the log and csv files as attached.

1. 32-bit cebreak:
[image]
[image]

2. 64-bit cebreak:
[image]
[image]

cebreak-64-bit log files

[spike_cebreak-01_64_bit.csv](https://github.com/openhwgroup/core-v-verif/files/9146777/spike_cebreak-01_64_bit.csv)
[spike_cebreak-01_64_bit.log](https://github.com/openhwgroup/core-v-verif/files/9146778/spike_cebreak-01_64_bit.log)
[veri_cebreak-01_64_bit.csv](https://github.com/openhwgroup/core-v-verif/files/9146779/veri_cebreak-01_64_bit.csv)
[veri_cebreak-01_64_bit.log](https://github.com/openhwgroup/core-v-verif/files/9146780/veri_cebreak-01_64_bit.log)

cebreak-32-bit log files

[32-bit_veri_cebreak-01.csv](https://github.com/openhwgroup/core-v-verif/files/9146787/32-bit_veri_cebreak-01.csv)
[32-bit_veri_cebreak-01.log](https://github.com/openhwgroup/core-v-verif/files/9146788/32-bit_veri_cebreak-01.log)
[32-bit_spike_cebreak-01.csv](https://github.com/openhwgroup/core-v-verif/files/9146789/32-bit_spike_cebreak-01.csv)
[32-bit_spike_cebreak-01.log](https://github.com/openhwgroup/core-v-verif/files/9146790/32-bit_spike_cebreak-01.log)

spidugu444, 2023-09-21:
In this it has an issue with MTVEC default value, on CVA6 RTL it is reading default value as 0x10040 on spike end it is reading 0x0.
please find similar issue here https://github.com/openhwgroup/cva6/issues/1312.

zchamski, 2025-01-09:
Can be closed as duplicate of https://github.com/openhwgroup/cva6/issues/1312.

closed 2025-01-16 by Gchauvon