openhwfoundation/cva6 #1418: CVXIF : wrong X_NUM_RS assign

AyoubJalali, 2023-09-14:
As we discuss @MikeOpenHWGroup on how can the core behave in the this case : 
When the cva6 send a request to execute a custom instruction with 3 source register, but X_NUM_RS=2.
should raise an exception or should the CVXIF agent execute the instruction just with 2 source register.

the question is : how can the core  behave in this case ?

Gchauvon, 2023-09-15:
You should just raise an illegal exception since executing a 3 source operands instruction should not be possible with X_NUM_RS=3

AyoubJalali, 2023-09-15:
> The behaviour doesn't depends on the core in this case but only on the coprocessor connected which can accept and execute the instruction or not accept and let the core raise an illegal instruction.
> 
> Anyway, I believe that raising an illegal exception make more sense in this case for a coprocessor since CVA6 would only provide 2 source operands.

Yes that's right.
Can we set X_NUM_RS to 3 ? In that case everything work great

Gchauvon, 2023-09-15:
No we can't, it is unnecessary since no coprocessor use 3 source register yet. And it has a considerable impact on area by adding lots of gate.

AyoubJalali, 2023-09-15:
> No we can't, it is unnecessary since no coprocessor use 3 source register yet. And it has a considerable impact on area by adding lots of gate.

ok understood

AyoubJalali, 2023-09-15:
Base on the discussion I'll close this issue
@Gchauvon Thanks

closed 2023-09-15 by AyoubJalali