openhwfoundation/cva6 #1419: CVA6 Makefile sets random seed to "1"

MikeOpenHWGroup, 2023-09-14:
The Makefile at `core-v-verif/cva6/sim/Makefile` "hardwires" the random seed:
```
ALL_SIMV_UVM_FLAGS      = -licwait 20 -l +ntb_random_seed=1 \
```
I see that `core-v-verif/cva6/sim/cva6-simulator.yaml` _appears_ to offer the ability to override this:
```
<out>/vcs_simv +vcs+lic+wait gen="true" <sim_opts> +ntb_random_seed=<seed> <cov_opts>
```
How does this work?  Once the Makefile sets `+ntb_random_seed` to "1", can it be over-ridden by the configuration in the yaml?

AyoubJalali, 2023-11-24:
Hello @MikeOpenHWGroup indeed **_+ntb_random_seed_** it override but not by **core-v-verif/cva6/sim/cva6-simulator.yaml**, it's the --sv_seed option in **_verif/sim/cva6.py_**

JeanRochCoulon, 2025-01-02:
@MikeOpenHWGroup We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case?

MikeOpenHWGroup, 2025-01-03:
Thanks for the gentle reminder @JeanRochCoulon, I should have looked at this long ago.

I can see that starting on [line 1098 of cva6.py](https://github.com/openhwgroup/cva6/blob/86a80f0eaa4ddbe79360a817f3fd332c6a8e3c09/verif/sim/cva6.py#L1098) there is code to process command-line arguments `sv_seed` and `gen_sv_seed`.

All good!

closed 2025-01-03 by MikeOpenHWGroup