openhwfoundation/cva6 #1420: CVXIF : Wrong dualread length

AyoubJalali, 2023-09-14:
Hello,
I notice that in [cvxif implementation in cva6](https://github.com/openhwgroup/cva6/blob/master/core/include/cvxif_pkg.sv), has been declare the **dualread** signal in issue response as a 1 bit signal, instead of declare it as 3 bits signal base on the [cvxif spec](https://github.com/openhwgroup/core-v-xif/blob/main/docs/source/x_ext.rst)

Gchauvon, 2023-09-15:
It is an unsupported 32 bits feature. Implementation has been done when the dualread and the dualwrite signals were 1 bit signal.

Generally speaking, CVXIF is not up to date to its latest specification on the CVA6.

Gchauvon, 2023-12-06:
Feature not supported. A fix might come with an updated implementation of the CVXIF specification which is still in developement.

closed 2023-12-06 by Gchauvon