openhwfoundation/cva6 #1421: Reading mcycle CSR makes simulation fail

cathales, 2023-09-14:
This issue is about CVA6 end of test.

* Functionally incorrect behavior
* Confusing or extraneous status or error messages

Using `setStats()` in a program enables a counter. After `main()` returns, the counter is printed via a syscall. **This syscall fails**: the simulation times out and no `core_main.log` is produced (only `core_main.log.iss`).

A workaround is suggested in https://github.com/openhwgroup/core-v-verif/pull/1757 and using it (`-DNOPRINT`) makes the simulation succeed, producing the `core_main.log` file.

### Steps to Reproduce

1. URL to branch that exhibits the issue: [cathales/core-v-verif@failing-coremark](https://github.com/cathales/core-v-verif/tree/failing-coremark)
2. Command line: `bash cva6/regress/coremark.sh`
3. Logfile and/or wave-dump info (screen shots can be useful):

Output:

```
Thu, 30 Mar 2023 14:57:30 ERROR    ERROR return code: 1/2, cmd:make veri-testharness target=gen32 variant=rv32imac elf=/home/callart/core-v-verif/cva6/sim/out_2023-03-30/directed_c_tests/core_main.o path_var=/home/callart/core-v-verif/core-v-cores/cva6 tool_path=/shares/common/tools/spike/bin isscomp_opts="+define+WT_DCACHE+RVFI_TRACE+RVFI_MEM" issrun_opts="+debug_disable=1 +time_out=3000000 " isspostrun_opts="0x0000000080000000" log=/home/callart/core-v-verif/cva6/sim/out_2023-03-30/veri-testharness_sim/core_main.log &> /home/callart/core-v-verif/cva6/sim/out_2023-03-30/veri-testharness_sim/core_main.log.iss
```

End of `core_main.log.iss`:

```
Listening on port 38101
/home/callart/core-v-verif/cva6/sim/out_2023-03-30/directed_c_tests/core_main.o *** FAILED *** (tohost = 2147483647) after 3000012 cycles
CPU time used: 261036.36 ms
Wall clock time passed: 261106.24 ms
make: *** [Makefile:120: veri-testharness] Error 255
```

Cc @zchamski

JeanRochCoulon, 2025-01-02:
@cathales We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case?

cathales, 2025-01-06:
If we remove the `-DNOPRINT` workaround on CoreMark, we notice that the issue is still here.

So we do have a workaround for CoreMark, but the root issue is not fixed for other programs.