openhwfoundation/cva6 #1422: Jump to no aligned address

AyoubJalali, 2023-09-14:
Actually this not a bug, but it's a question about what should the CVA6 core do in this situation : 

- If the core does not support the C extension, and we try to jump to an address not aligned, because we can't provide a test case when the C extension is enabled, because of the definition of the jump instruction JALR that clean the LSB of the address target, so how should the core react  in this situation.

zarubaf, 2023-10-20:
Any updated? Don't we trigger this in the regression?

JeanRochCoulon, 2023-10-20:
@AyoubJalali   Where is the best place to log this point to be verified ?   ISACOV or another DVPlan ?   For sure, the behaviour need to be documented in design document because it seems to be cva6-specific.

@ASintzoff

MikeOpenHWGroup, 2023-11-07:
@AyoubJalali asked the question:
>  I'm talking about disabling the C extension, in that case the address should be 4-byte aligned boundary, and this a case to verify or at least know what should do.

In section 1.5, `Base Instruction-Length Encoding` of the [Unpriviledged ISA](https://drive.google.com/file/d/1s0lZxUZaa7eV_O0_WsZzaurFLLww7ou5/view) it says:
```
We use the term IALIGN (measured in bits) to refer to the instruction-address alignment constraint
the implementation enforces. IALIGN is 32 bits in the base ISA, but some ISA extensions, including
the compressed ISA extension, relax IALIGN to 16 bits. IALIGN may not take on any value other
than 16 or 32.
```
My read of this:
- For the `base ISA` IALIGN is 32-bits and instructions are constrained to align to 32-bit boundaries.
- For the `C ISA` IALIGN is 16-bits and instructions are constrained to align to 16-bit boundaries.

Furthermore, section 2.2, `Base Instruction Formats` says:
```
An instruction-address-misaligned exception is generated on a taken branch or unconditional jump
if the target address is not four-byte aligned. This exception is reported on the branch or jump
instruction, not on the target instruction.
```

So, to answer @AyoubJalali's question:
1. When the `C ISA` is disabled, all instructions must align on 32-bit (4-byte) boundaries.  (This is true for both 64-bit and 32-bit XLEN cores.)
2. If program attempts to branch or jump to an address that is not 32-bit aligned, the core must throw an instruction-address-misaligned exception.
3. Yes, we must verify this.  That is, for configurations of CVA6 in which the C ISA is not supported, we must attempt to jump to a non-32-bit-aligned address and check to see if the core throws an instruction-address-misaligned exception _on the branch or jump instruction_.

MikeOpenHWGroup, 2023-11-07:
Hi @AyoubJalali, if you agree with my understanding of ISA and what we must do about it, then I think this Issue can be closed.  (We should probably create a new Task Issue to ensure we actually test it.)

AyoubJalali, 2023-11-08:
> Hi @AyoubJalali, if you agree with my understanding of ISA and what we must do about it, then I think this Issue can be closed. (We should probably create a new Task Issue to ensure we actually test it.)

totally agree

closed 2023-11-08 by AyoubJalali