openhwfoundation/cva6 #1425: Rework syscalls.c once system prints are implemented

cathales, 2023-09-14:
[#1857](https://github.com/openhwgroup/core-v-verif/pull/1857/files#diff-62f67c1d16e0eb8da38c8838b66b69e31890c555b9d2ad5509d245754da1f25d) introduced two new functions:

https://github.com/openhwgroup/core-v-verif/blob/1eac965d073abc18ab6aff5e6f62fd635b060746/cva6/tests/custom/common/syscalls.c#L78-L81

and

https://github.com/openhwgroup/core-v-verif/blob/1eac965d073abc18ab6aff5e6f62fd635b060746/cva6/tests/custom/common/syscalls.c#L164-L173

We might want to move or modify them once system prints is made via UART both in Spike and CVA6. (Related: openhwgroup/cva6#1421)

cathales, 2023-09-14:
Fixed by https://github.com/openhwgroup/cva6/pull/1395

zchamski, 2024-02-15:
This is a topic still open as the main focus remains on verification, leaving core-environment interactions at lower priority.