openhwfoundation/cva6 #1427: Spike : Unrecognized CSR

AyoubJalali, 2023-09-14:
Hello,
As i know the ICACHE_CSR is implemented in the STEP1 configuration, so i generate a test with a csr instruction with ICACHE_CSR = 12'b700, the spike model raised an exception, because it didn't recognize the CSR (i don't think it's implemented), but the CVA6 did not raised an exception.

to reproduce the bug : 

- Add the yaml description of the test in [cva6_base_testlist.yaml](https://github.com/openhwgroup/core-v-verif/blob/master/cva6/sim/cva6_base_testlist.yaml)

- test: riscv_rand_jump_illegal_test
  description: >
    Jump among large number of sub-programs, stress testing iTLB operations.
  gen_opts: >
    +instr_cnt=300
    +num_of_sub_program=0
    +directed_instr_0=riscv_load_store_rand_instr_stream,10
    +directed_instr_1=riscv_jal_instr,20
    +illegal_instr_ratio=100
    +tvec_alignment=8
  iterations: 2
  gen_test: cva6_instr_base_test_c
  rtl_test: core_base_test

and run the commend :

python3 cva6.py --testlist=cva6_base_testlist.yaml --test riscv_rand_jump_illegal_test --iss_yaml cva6.yaml --target cv32a60x --iss=vcs-uvm,spike --simulator_yaml ../env/corev-dv/simulator.yaml --iss_timeout 300 -i 1 --seed 1472969439

with these version : 

1. core-v-verif -> branch : cva6/dev
2. core-v-core/cva6 -> branch : master
3. sim/dv (riscv-dv) -> branch : 0ce85d3

zchamski, 2023-12-14:
I am developing a proof-of-concept solution to add arbitrary CSRs via Spike extensions (`spike --extension=cv32a6embedded ...`).  This could be the generic approach to *add* CSRs as apposed to disabling unavailable ones (see #1599, #1145).

MarioOpenHWGroup, 2024-01-08:
Already merged by @zchamski  https://github.com/openhwgroup/core-v-verif/pull/2334

zchamski, 2024-04-04:
The `ICACHE` and `DCACHE` CSRs are now supported through Spike extension `cv32a60x`.  While the extension may get renamed (currently it only provides the two CSRs mentioned), I propose to close this issue.

closed 2024-04-04 by AyoubJalali