openhwfoundation/cva6 #1432: Environent setup error

hhhsiang, 2023-09-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I installed the necessary RISC-V tools according to the [README.md](https://github.com/openhwgroup/cva6#readme)
After executing the following commands:
`export DV_SIMULATORS=veri-testharness,spike`
`bash verif/regress/smoke-tests.sh`
I encountered the following errors:
[image]
[image]
[image]

I also ran the following command:
`bash verif/regress/dv-riscv-compliance.sh`
However, it still displayed the same error:
[image]

How could I fix it?
Thanks in advance!

oliverbm67, 2023-09-24:
For the first issue "ERROR: sphinx-rtd-theme [...] " and "ERROR: python-jsonschema" , these are python package issue. It reads like you have version of these packages that are more recent than the requirements. To avoid messing anything in your system, I would suggest trying to setup a python virtual env ( see  for example https://docs.python.org/3/library/venv.html but you have plenty of resources online).

For "device-tree-compiler not found", you need to install device tree compiler. It is often available in your linux distro package repository under the name device-tree-compiler ( I think it is called dtc on RHEL/Fedora systems).

For the other errors the screenshot does not tell much. I would suggest you fix the previous errors I mentioned and try again. If you still get errors at the make veri-testharness step, please attach the log files to this issue ( these are the files called add.log and add.log.iss in one of the screenshot).

zarubaf, 2023-10-23:
Thanks @oliverbm67 for the very helpful answer. Hope that solves your issue @hhhsiang.

closed 2023-10-23 by zarubaf