openhwfoundation/cva6 #1434: [TASK] Generate CSR DVplan

JeanRochCoulon, 2023-09-18:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Generate the Dvplan for CSRs in embedded config: field access, illegal CSRs

### Required Changes

Updating STEP1 Dvplan

### Current Status

STEP1 dvplan 

### Risks

none

### Prerequisites

CSR Spec for embedded

### KPI (KEY Performance Indicators)

none

### Description of Done

CSR Dvplan file for embedded config

JeanRochCoulon, 2023-10-10:
is this task started ?

AnouarZajni, 2023-10-10:
I started it today.

AnouarZajni, 2023-10-13:
@JeanRochCoulon, task done waiting for merge approval.

JeanRochCoulon, 2023-10-13:
Can you refer the PR number ?

AnouarZajni, 2023-10-13:
CSR DVPLAN embedded is PR !86
I've also updated hvp and RAL files to achieve dvplan goals in !87

AnouarZajni, 2023-10-18:
@ASintzoff, updated CSR Dvplan and pushed to the same merge request !86

AnouarZajni, 2023-10-18:
@ASintzoff , RAL files updated to reflect Dvplan with Legal/Illegal values and CSR counter features. Pushed to 87

AnouarZajni, 2023-10-25:
@ASintzoff , Updated merge !86 following feedbacks

AnouarZajni, 2023-11-09:
Merge Done, closing the the task

JeanRochCoulon, 2023-11-30:
@MikeOpenHWGroup  Can I ask you to review this task ? If you think this task is completed, please change the status from "Done" to "finished". Thanks

https://github.com/openhwgroup/cva6/tree/master/verif/docs/VerifPlans/csr_access

MikeOpenHWGroup, 2023-11-30:
I am re-opening this Issue as @JeanRochCoulon has requested a review of the CSR DV Plan.

MikeOpenHWGroup, 2023-12-11:
I have conducted a review of the CSR DVPlan, see #1697.

JeanRochCoulon, 2024-05-22:
@AnouarZajni @ASintzoff  Is this task completed ?   If yes, please refer the related PR.

AnouarZajni, 2024-05-22:
@JeanRochCoulon , yes task is done in MR !85 with PR #1606

closed 2024-05-22 by JeanRochCoulon