openhwfoundation/cva6 #1436: [TASK] Report embedded code coverage in dashboard

JeanRochCoulon, 2023-09-18:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Report Embedded code coverage in dashboard

### Required Changes

Create a graph in the dashboard to follow the code coverage evolution, as it is done for kgates. Report statement, condition, internal toggle in absolute

### Current Status

todo

### Risks

none

### Prerequisites

Give access to the CI logs to allow failing test debugging

### KPI (KEY Performance Indicators)

none

### Description of Done

Graph in the dashboard to indicate the code coverage score

AEzzejjari, 2023-10-09:
T1: Display the coverage score in the console at the end of the code-coverage job. (Done)

AEzzejjari, 2023-10-17:
Script to display code coverage score and store it in yaml format is PR #1526

cathales, 2023-10-19:
- As mentioned above, "Code coverage report" job now generates a report to be displayed in the Dashboard thanks to https://github.com/openhwgroup/cva6/pull/1526
- Scheduled *verif* pipeline to run every Sunday, however some *verif* jobs may be timing out. It will be possible to fix that later. These *verif* pipelines run the "code coverage report" job
- A *coverage* graph will be added when we have several values of code coverage.

cathales, 2023-10-23:
The first result is available, it is rendered as below:

[image]

cathales, 2023-11-09:
We have several results now, I'll add a graph for coverage by the end of the week.

cathales, 2023-11-09:
Done (you might need to refresh the page)

https://riscv-ci.pages.thales-invia.fr/dashboard/index.html

closed 2023-11-09 by cathales

JeanRochCoulon, 2023-11-30:
Thanks @cathales for this work. As CC has been integrated to the dashboard, it is user friendly to improve CC, the task is completed.