openhwfoundation/cva6 #1437: [TASK] Condition RTL with parameters  (MMU, FPU, …)

JeanRochCoulon, 2023-09-18:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Condition RTL with parameters  (MMU, FPU, …)

### Required Changes

Adding if(CVA6Cfg.*) in RTL to identify the dead code and remove the dead gates from netlist. Based on if() directives, VCS is able to identify the unused code, this will allow to reach 100% code coverage.

### Current Status

In progress

### Risks

none

### Prerequisites

none

### KPI (KEY Performance Indicators)

100% code coverage

### Description of Done

have a configurable RTL

AEzzejjari, 2023-10-10:
T1: Condition RTL with FPU parameter. (Done)
T2: Condition RTL with BItManip parameter. (Done)
T3: Condition RTL with MMU parameter. (Done)
T4: Condition RTL with AMO parameter. (In progress)

JeanRochCoulon, 2023-10-19:
Status of CC on 18/10/203:
[['COMPONENT', 'SCORE', 'LINE', 'COND', 'TOGGLE'],
 ['i_cva6', 75.32, 85.17, 82.36, 58.43],
 ['commit_stage_i', 75.92, 96.61, 60.0, 71.15],
 ['controller_i', 76.58, 89.09, 70.0, 70.65],
 ['csr_regfile_i', 42.01, 63.07, 39.74, 23.22],
 ['ex_stage_i', 77.55, 87.96, 88.33, 56.35],
 ['gen_cache_wt.i_cache_subsystem', 74.09, 85.75, 80.3, 56.22],
 ['i_frontend', 89.88, 97.18, 91.14, 81.33],
 ['id_stage_i', 65.48, 86.21, 56.16, 54.06],
 ['issue_stage_i', 87.09, 95.67, 88.66, 76.93]]

JeanRochCoulon, 2023-10-20:
After AMO modification:

| COMPONENT                           | SCORE | LINE  | COND  | TOGGLE |
|------------------------------------|-------|-------|-------|--------|
|i_cva6| 75.32| 85.58| 83.16| 57.21|
|commit_stage_i| 68.87| 96.61| 61.54| 48.46|
|controller_i| 79.24| 89.29| 77.78| 70.65|
|csr_regfile_i| 40.4| 61.41| 38.46| 21.32|
|ex_stage_i| 78.09| 89.59| 90.36| 54.32|
|gen_cache_wt.i_cache_subsystem| 75.09| 88.31| 81.69| 55.28|
|i_frontend| 89.56| 97.18| 91.14| 80.36|
|id_stage_i| 61.77| 78.45| 53.42| 53.45|
|issue_stage_i| 86.67| 95.19| 88.35| 76.49|

JeanRochCoulon, 2023-10-22:
@AEzzejjari in decoder, the bit manip instructions are not all covered, is a bitmanip test missing ?  
[image]

fatimasaleem, 2023-10-23:
@JeanRochCoulon I just added the bitmanip arch-tests for this configuration as well, by any chance we miss running those?

JeanRochCoulon, 2023-10-23:
Hello @AyoubJalali @AEzzejjari @fatimasaleem @ASintzoff 
Of course the new arch-test tests testing bit manip will be integrated to code coverage job. But before I would be happy to understand why the generated tests do not generate all bit manip instructions.

AyoubJalali, 2023-10-23:
> Hello @AyoubJalali @AEzzejjari @fatimasaleem @ASintzoff Of course the new arch-test tests testing bit manip will be integrated to code coverage job. But before I would be happy to understand why the generated tests do not generate all bit manip instructions.

simply because I didn't enabled them on the tests' generation, why ? -> AS YOU KNOW the bitmanip have some fraction extension zb*, so you wanna generate all bitmanip instructions supported by RISCV-DV, or you have something specific.

BITMANIP instruction supported by RISCV-DV (version  **_0ce85d3_**) are :
https://gitlab-tss.gemalto.com/riscv/riscv-dv/-/blob/timeoutfix/src/isa/rv32b_instr.sv?ref_type=heads

fatimasaleem, 2023-10-23:
Hi @AyoubJalali CVA6 supports all Bitmanip sub-extensions.

AyoubJalali, 2023-10-23:
> Hi @AyoubJalali CVA6 supports all Bitmanip sub-extensions.

Yes but RISCV-DV generate some instructions that hasn't implemented 
check what riscv-dv support for [bitmanip](https://github.com/chipsalliance/riscv-dv/blob/master/src/isa/rv32b_instr.sv)

JeanRochCoulon, 2023-10-24:
After disabling the dcache_ctrl module:

| COMPONENT                           | SCORE | LINE  | COND  | TOGGLE |
|------------------------------------|-------|-------|-------|--------|
| i_cva6| 76.27| 87.39| 84.02| 57.39|
| commit_stage_i| 68.87| 96.61| 61.54| 48.46|
| controller_i| 79.24| 89.29| 77.78| 70.65|
| csr_regfile_i| 43.46| 67.43| 39.74| 23.22|
| ex_stage_i| 78.15| 89.71| 90.36| 54.38|
| gen_cache_wt.i_cache_subsystem| 76.4| 90.35| 83.49| 55.35|
| i_frontend| 89.56| 97.18| 91.14| 80.36|
| id_stage_i| 62.48| 79.96| 53.42| 54.06|
| issue_stage_i| 87.24| 96.14| 88.66| 76.93|

JeanRochCoulon, 2023-10-27:
After disabling SMODE, the CC is improved but not so much because CSR test does not cover very much the CSR module:

| COMPONENT                           | SCORE | LINE  | COND  | TOGGLE |
|------------------------------------|-------|-------|-------|--------|
| i_cva6| 76.59| 87.48| 84.87| 57.43|
| commit_stage_i| 68.87| 96.61| 61.54| 48.46|
| controller_i| 73.08| 89.29| 77.78| 52.17|
| csr_regfile_i| 45.56| 65.57| 48.33| 22.78|
| ex_stage_i| 77.59| 89.56| 90.36| 52.86|
| gen_cache_wt.i_cache_subsystem| 76.74| 90.35| 83.03| 56.84|
| i_frontend| 88.92| 97.18| 91.14| 78.43|
| id_stage_i| 66.07| 81.54| 64.06| 52.62|
| issue_stage_i| 86.16| 96.14| 88.66| 73.7|

AyoubJalali, 2023-10-27:
> After disabling SMODE, the CC is improved but not so much because CSR test limits teh coverage:
> 
> COMPONENT	SCORE	LINE	COND	TOGGLE
> i_cva6	76.59	87.48	84.87	57.43
> commit_stage_i	68.87	96.61	61.54	48.46
> controller_i	73.08	89.29	77.78	52.17
> csr_regfile_i	45.56	65.57	48.33	22.78
> ex_stage_i	77.59	89.56	90.36	52.86
> gen_cache_wt.i_cache_subsystem	76.74	90.35	83.03	56.84
> i_frontend	88.92	97.18	91.14	78.43
> id_stage_i	66.07	81.54	64.06	52.62
> issue_stage_i	86.16	96.14	88.66	73.7

I have some leads to improve the csr module, it's on going

JeanRochCoulon, 2023-10-31:
Hello @AEzzejjari 
sfence_vma is a supervisor instruction. This instruction can also be conditioned by CVA6Cfg.RVS. Dedicated code is implemented in controller and decoder modules.

JeanRochCoulon, 2023-11-07:
Thanks to Rabat team and @ASintzoff,  the CC has improved a lot:

| COMPONENT                           | SCORE | LINE  | COND  | TOGGLE |
|------------------------------------|-------|-------|-------|--------|
|i_cva6| 80.67| 94.29| 87.09| 60.6|
|commit_stage_i| 70.28| 100.0| 61.54| 49.3|
|controller_i| 87.14| 100.0| 87.5| 73.91|
|csr_regfile_i| 67.29| 98.08| 58.97| 44.8|
|ex_stage_i| 82.7| 94.54| 93.7| 59.86|
|gen_cache_wt.i_cache_subsystem| 78.89| 93.06| 85.38| 58.24|
|i_frontend| 89.37| 97.53| 89.87| 80.7|
|id_stage_i| 71.39| 91.17| 69.09| 53.9|
|issue_stage_i| 87.28| 96.14| 88.66| 77.05|

AEzzejjari, 2023-11-09:
T5: Condition RTL with Supervisor mode parameter.
T6: Condition RTL with User mode parameter.
T7: Condition RTL with Debug mode parameter.

AyoubJalali, 2023-11-09:
> Thanks to Rabat team and @ASintzoff, the CC has improved a lot:
>
> COMPONENT	SCORE	LINE	COND	TOGGLE
> i_cva6	80.67	94.29	87.09	60.6
> commit_stage_i	70.28	100.0	61.54	49.3
> controller_i	87.14	100.0	87.5 [...]

[['COMPONENT', 'SCORE', 'LINE', 'COND', 'TOGGLE'],
 ['i_cva6', 81.29, 94.72, 87.81, 61.35],
 ['commit_stage_i', 70.0, 100.0, 61.54, 48.46],
 ['controller_i', 87.14, 100.0, 87.5, 73.91],
 ['csr_regfile_i', 67.2, 98.08, 58.97, 44.53],
 ['ex_stage_i', 82.7, 94.54, 93.7, 59.86],
 ['gen_cache_wt.i_cache_subsystem', 79.5, 93.06, 85.96, 59.48],
 ['i_frontend', 89.79, 97.53, 91.14, 80.7],
 ['id_stage_i', 78.12, 95.01, 85.45, 53.9],
 ['issue_stage_i', 87.28, 96.14, 88.66, 77.05]]

@JeanRochCoulon I added some tests that improve id_stage (decoder & compressed_decoder) by more than 10%

JeanRochCoulon, 2024-05-19:
Hello @AEzzejjari,    THis task is a continuous improvment. But if you consider that the main work is done, I would be agree to close it.  Tell me !

AEzzejjari, 2024-05-20:
I agree with you, the main work is done, so I think it's time to close this task.

closed 2024-05-22 by JeanRochCoulon